schreibfaul1/ESP32-audioI2S #142: Streams stuttering on start with PSRAM enabled.

CelliesProjects, 2021-04-06:
Hi Wolle. Another issue.

As the title suggest, (some) streams are stuttering for 2-3 seconds but not all of them. It also happens on (some) files. In the example are two problematic streams. All the problematic streams play fine without psram enabled.

I observed this issue before, but since [this commit to esp32 arduino](https://github.com/espressif/arduino-esp32/commit/5502879a5b25e5fff84a7058f448be481c0a1f73) I have so little ram left after boot that a psram buffer is unavoidable.

Could you please take a look?

```c++
#include <Audio.h>  /* https://github.com/schreibfaul1/ESP32-audioI2S */

#define I2S_BCK     21
#define I2S_WS      26
#define I2S_DOUT    22

Audio audio;

void setup() {
  audio.setPinout(I2S_BCK, I2S_WS, I2S_DOUT);
  WiFi.begin("xxx", "xxx");
  while (!WiFi.isConnected()) {
    delay(10);
  }
  ESP_LOGI(TAG, "Connected. Starting MP3...");
  //audio.connecttohost("http://icecast.omroep.nl/radio1-bb-mp3");
  audio.connecttohost("http://icecast.omroep.nl/3fm-bb-mp3");
  audio.setVolume(20);
}

void loop() {
  audio.loop();
}
```

schreibfaul1, 2021-04-06:
Hi Cellie,
it is how you write. With the new version, SRAM is reduced.
V1.0.6 ESPIDF 3.3.5    188216 bytes free heap
V2.0.0 ESPIDF 4.4       150999 bytes free heap
PSRAM has a queue as a bottleneck. When the queue (16KB ?)  is full, the speed is halved to 40MHz. That's why SRAM is the better choice. If PSRAM is available, the buffers of the decoders and the input buffer are moved there.

CelliesProjects, 2021-04-07:
That's clear. For now I just stay on the 3.3 branch and see if the 4.2 branch gets any leaner. 
Thanks again for the good work.

closed 2021-04-07 by CelliesProjects