schreibfaul1/ESP32-audioI2S #145: Mp3 play stutters

grch87, 2021-04-10:
Hi Wolle,
I discovered that when using the latest version the mp3 play is stuttering (not checked with AAC or FLAC). I reverted back to https://github.com/schreibfaul1/ESP32-audioI2S/commit/8733a82d68ed89422b2389cbb52fe36bdbb72682
and it works perfectly there.

I'm using your great library in the Espuino Software with a lolin d32 pro.

Regards
Christoph

schreibfaul1, 2021-04-10:
Hi Christoph,
as of this version, the buffer from the AAC and MP3 decoder is relocated to the PSRAM if possible. PSRAM is a little slower than SRAM. With IDF version 4.4. which will be published in the next few days (with the Arduino ESP32 version 2.0.0), otherwise it can get tight. FLAC uses too much heap and can therefore only run in PSRAM. It's a tradeoff between speed and storage.

schreibfaul1, 2021-04-10:
Hi Christoph, 
I've now changed the order. Heap is used as long as there is enough memory. If not, a search is made for PSRAM. This means that stations are played more smoothly.
According to my measurements, loops are okay for up to 30 ms. Disturbances occur above this. With PSRAM I measured about 26ms. Maybe that's not the case with all streams.
Best regards from Berlin
Wolle

closed 2021-06-16 by stale[bot]