schreibfaul1/ESP32-audioI2S #152: No matching function for call to 'Audio::connecttohost(String&)

bhpjean, 2021-04-16:
I get this error while compiling. 

Audio::connecttohost(const char*, const char*, const char*)
     bool connecttohost(const char* host, const char* user = "", const char* pwd = "");

error: no matching function for call to 'Audio::connecttoFS(fs::SDFS&, String&)'
                                                                  audio.connecttoFS(SD, url); 

How can I fix this problem?
In previous versions, this error does not appear.

schreibfaul1, 2021-04-16:
Hello,
String has been replaced by const char*,
try this:
```
String station = "http://stream.friskyradio.com:9000/frisky_mp3_h";
audio.connecttohost(station.c_str());
```

bhpjean, 2021-04-17:
Ok. Thank you.

However, another problem arose ...
It cannot play mp3 and FLAC files from an SD card.
An error crashes on the monitor serial
/mp3/02equinoxe.mp3

audio_info: buffers freed, free Heap: 166452 bytes
audio_info: Reading file: "mp3 / 02equinoxe.mp3
"
audio_info: The mp3
  format is not supported 

audio_info: buffers freed, free Heap: 166096 bytes
audio_info: Reading file: "MikeMareen/01.flac
"
audio_info: The flac
 format is not supported

bhpjean, 2021-04-25:
Using the software on this page, I converted the ST-610 analog tuner into internet radio.
Thanks to schreibfaul1 for all the hard work put into the ESP32-audioI2S project.

a short demonstration of how the converted Technics ST-610 works.

https://youtu.be/nWQcmNH12Pc

schreibfaul1, 2021-04-25:
This is a nice project, thanks for sharing!

kamil-m-m, 2021-04-30:
> Using the software on this page, I converted the ST-610 analog tuner into internet radio.
> Thanks to schreibfaul1 for all the hard work put into the ESP32-audioI2S project.
> 
> a short demonstration of how the converted Technics ST-610 works.
> 
> https://youtu.be/nWQcmNH12Pc
> &
> Harman/Kardon HK3300
> https://youtu.be/sKnqNAjZ6XY

Very good project. Any details?
Sterowanie ESP32 pilotem, czy wykorzystałeś istniejące przyciski od Technicsa?

Pozdrawiam
Kamil

bhpjean, 2021-04-30:
> Very good project. Any details?
> Sterowanie ESP32 pilotem, czy wykorzystałeś istniejące przyciski od Technicsa?
>
> Pozdrawiam
> Kamil

obydwa rozwiązania działają , tzn przyciski na tunerze oraz oryginalny pilot technics.
do obsługi klawiatury wykorzystałem Moduł PCF8574 - 8 bitowy port sterowany I2C
oraz bibliotekę ze strony https://diyi0t.com/keypad-arduino-esp8266-esp32/
oryginalna klawiatura tunera ST-610 ( 4 x 5 ) została nieco zmodyfikowana i dostosowana
pod klawiaturę 4 x 4 . Całość t.j. klawiatura i wyświetlacz chodzą po wspólnej szynie i2c
co ograniczyło ilość przewodów pomiędzy kontrolerem a panelem przednim.

Pozdrawiam

Piotr

closed 2021-07-09 by stale[bot]