schreibfaul1/ESP32-audioI2S #157: using SD card of TFT  ST7735

optimaxch, 2021-05-04:
hi schreibfaul,

using a 128x128 ST7735  TFT  to load image from SD card,
 i like to implement your audio code to play mp3 from this SD card.

cannot merge  adafruit SdFat library...

#include "Adafruit_ST7735.h"  // Adafruit TFT 128x128
#include "Adafruit_GFX.h"     // Core graphics library
#include "SPI.h"

#include "SdFat.h"                // SD card & FAT filesystem library
#include "Adafruit_SPIFlash.h"    // SPI / QSPI flash library
#include "Adafruit_ImageReader.h" // Image-reading functions

can you help ?

geht das überhaupt?
danke aus villach
  rudi

[image]

schreibfaul1, 2021-05-04:
Hallo,
SDFat wird nicht unterstützt, nur SD und SD_MMC. Es gab schon Versuche SDFat zu implementieren, aber das war nicht in allen Fällen erfolgreich. Es gibt ein Issue https://github.com/schreibfaul1/ESP32-audioI2S/issues/111 in dem anp59 eine Möglichkeit zur Realisierung vorgestellt hat. Wenn SDFat installiert ist, funktioniert das.

Viele Grüße aus Berlin
Wolle

optimaxch, 2021-05-06:
hi wolle,
danke,  anp59  works well for audio, 
( audio von sdcard  u stream OK )
aber meine skills sind too low um ein bild von der sd card einzulesen (Adafruit_ImageReader) und am tft anzuzeigen.
kannst du das interpretieren ?

danke für deine mühe
alles gute

rudi

[image]

schreibfaul1, 2021-05-06:
Dazu kann ich nichts sagen, ich kenne weder das Display noch die Adafruit Library. Ich kann die Lib von Bodmer https://github.com/Bodmer/TFT_eSPI empfehlen, damit habe ich gute Erfahrungen gemacht.

optimaxch, 2021-05-06:
danke,
noch eine frage:
zuerst, wie kann ich dich auf ein bier einladen - oder muss ich warten bis du nach kärnten auf urlaub kommst ?

und:
audio soll als " add on " in meinem ESP32 pico  project laufen, welches hauptsächlich dazu da ist um etwa 1000 LEDs mit PCA9956 zu controllen.
d.h. in der loop läuft die   LED_control PCA9956.

wie ich das verstehe, braucht aber das audio die ganze loop.
  das würde heissen audio radio-streaming ist parallel zum LED_control nicht möglich.
  aber es wäre denkbar kurze   audio ansagen  ( etwa 3sec lang ) - von SDcard -   mit       if(!audio_eof_mp3)   auszuführen -  oder ?

danke für deine zeit
gruss aus villach
 rudi
[image]

anp59, 2021-05-07:
Hi optimaxch,
ohne Deinen Quellcode kann ich auch keine Hilfestellung geben. Die Instanz Adafruit_ImageReader muss mit einer Referenz auf ein Objekt des Typs FatFileSystem initialisiert werden. Im vorliegenden also mit SD (=SdFat). Wahrscheinlich ist auch eine Anpassung im Header Adafruit_Imagereader nötig. Ohne Quellcode von Dir kann ich jedoch auch nicht weiterhelfen. Sorry.

schreibfaul1, 2021-05-07:
Hi optimaxch,
die Audio Lib muss periodisch aufgerufen werden. Um Aussetzer zu vermeiden muss audio.loop() wenigstens alle 30ms durchlaufen werden. Es sei denn die AudioI2S Lib hat nichts zu tun, dann ist das egal. Im großen loop() selbst können noch andere Routinen stehen, sofern diese nur kurz laufen. Du kannst die audio.loop() auch in einem extra Task laufen lassen, zB. wie hier: https://github.com/mhier/LauraBox/blob/master/LauraBox.ino und schaue, wie _runAudioLoop_ aufgerufen wird.
vG
Wolle

closed 2021-07-15 by stale[bot]