schreibfaul1/ESP32-audioI2S #162: unicode2utf8 is not complete

en-ot, 2021-05-11:
Audio::unicode2utf8 can not convert all unicode characters to utf-8.
Japanese for example, and output is garbage in this case.

I think it's because there is a problem when output utf-8 character is 3 or more bytes long, and we are using the same buffer for input and output.
Can You please correct this?

The function that converts all unicode ranges is FsUtf::cpToMb for example.

schreibfaul1, 2021-05-12:
So far the converted part of the character set was limited to Europe, now I have used code from FsUtf :: cpToMb to represent everything that is contained in id3metadata.

en-ot, 2021-05-17:
New version is decoding UTF-8 correctly.
Also it is more stable than that I've used before - from anp59 project. It now can setFilePos without crashing time to time :)
Thank You very much!

closed 2021-05-17 by en-ot