schreibfaul1/ESP32-audioI2S #165: Volume control

centauro120, 2021-05-15:
hello,
I have built a loudspeaker in which esp controls the volume, unfortunately at the first volume level the music is too loud. Therefore, I have a question.
Is it possible to turn down the volume at the first volume level, but to make the music play 100% at the highest volume level?
Sorry for my English.
greetings.

nodoubtman, 2021-05-16:
> hello,
> I have built a loudspeaker in which esp controls the volume, unfortunately at the first volume level the music is too loud. Therefore, I have a question.
> Is it possible to turn down the [...]

audio.setVolume(21);

21 is the MAX VOLUME...

:o)

schreibfaul1, 2021-05-16:
22 values ​​from 0 ... 64 are entered in volumetable, which approximate a logarithmic curve. You can extend the range to 0 ... 255. In the routine Gain() then shift right 8 times instead of 6 times for both channels. This reduces the minimum volume by a factor of 4

aaga, 2021-06-02:
I made this exact change on my own fork of the library. My fork is currently out of date, but here is the relevant commit so you can see what the change looks like: https://github.com/aaga/ESP32-audioI2S/commit/0da11356db097c9bf643bff62ad7887badf849ed

centauro120, 2021-06-23:
Thank you very much for help, everything works.

closed 2021-06-23 by centauro120