schreibfaul1/ESP32-audioI2S #168: ESP32 pico    connecttospeech  hang

optimaxch, 2021-06-06:
after establishing SSL
[image]

schreibfaul1, 2021-06-06:
Thanks for the hint, Google TTS has changed the reaponse header, the parameter contentLength no longer exists. I have adjusted that, connecttospeech() works now.

optimaxch, 2021-06-07:
Thanks to **YOU,** for your excellent libs.
using  on ESP32-pico   w    vs-code-PIO  :      audio I2S    combined with     tft ILI9486
**ALL WORKING WELL**
[210607_tft+.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/6609178/210607_tft%2B.zip)

hope this can help someone to save time...

thanks
 opti

closed 2021-06-07 by optimaxch

optimaxch, 2022-04-09:
servus wolle,

vielleicht hilft dies jemandem ;
ich hab sehr lang gebraucht... - es funktioniert
( und ist es überhaupt richtig )
könnte in deine wiki  FAQ passen

  gruss aus villach
  rudi
[image: image.png]