schreibfaul1/ESP32-audioI2S #171: Just a question?

nodoubtman, 2021-06-22:
Hello. I just have one question.

What is the difference between audio I2S and vs1053 ?

thanks.
Marc.

biologist79, 2021-06-22:
i2s is a protocol whereas vs1053 is a chip.
When using I2S with this lib, ESP32 has to decode the music in software, makes i2s of it and send it to an external dac. vs1053 in contrast decodes music in hardware and ESP32 only tells vs1053 what to do :)

nodoubtman, 2021-06-22:
> 
> 
> i2s is a protocol whereas vs1053 is a chip.
> When using I2S with this lib, ESP32 has to decode the music in software, makes i2s of it and send it to an external dac. vs1053 in contrast decodes music in hardware and ESP32 only tells vs1053 what to do :)

Thank you for your time :)
good evening :)
Marc.

closed 2021-08-07 by nodoubtman