schreibfaul1/ESP32-audioI2S #172: Trying to added custom .mp3 url to read...

nodoubtman, 2021-06-29:
Hello, i'm trying to add a custom .mp3 url here's my code, but it plays only for 1 second..
can you help ?
Thanks.
Marc.

audio.connecttositemp3("clockshops.com", "/wp-content/uploads/2016/05/3-Mech_StMichael.mp3");

// Ajouté par Marc Paradis le 28 juin 2021
bool Audio::connecttositemp3(const char* host, const char* path) {
  setDefaults();
  bool     f_ct = false;
  uint32_t bytesCanBeWritten  = 0;
  uint32_t bytesCanBeRead     = 0;
  int32_t  bytesAddedToBuffer = 0;
  int16_t  bytesDecoded       = 0;
  uint32_t contentLength      = 0;

  //int httpCode;

<code>
  String resp =     String("GET ") + path + String(" HTTP/1.1\r\n")
                    + String("Host: ") + host + String("\r\n")
                    + String("User-Agent: Mozilla/5.0 \r\n")
                    + String("Accept-Encoding: identity\r\n")
                    + String("Accept: bytes\r\n")
                    + String("Connection: close\r\n\r\n");

  if (!clientsecure.connect(host, 443)) {
    log_e("Connection failed");
    return false;
  }

  log_e("Connection OK.");

  clientsecure.print(resp);
  sprintf(chbuf, "SSL has been established, free Heap: %u bytes", ESP.getFreeHeap());

  if (audio_info) audio_info(chbuf);

  /*
    Date: Tue, 29 Jun 2021 00:11:46 GMT
    Last-Modified: Fri, 20 May 2016 19:52:57 GMT
    Content-Length: 334464
    Content-Type: audio/mpeg
  */

  while (clientsecure.connected()) { // read the header
    String line = clientsecure.readStringUntil('\n');

    if (line.startsWith(F("Content-Length: "))) {
      contentLength = line.substring(15).toInt();

      char bufferContentLength[8];
      sprintf(bufferContentLength, "%d", contentLength);
      if (audio_info) audio_info(bufferContentLength);
    }

    if (line.startsWith("Content-Type")) {
      if (line.indexOf("audio/mpeg") > 12) f_ct = true; // Content-Type mpeg seen
    }

    line += "\n";
    if (line == "\r\n") break;
  }

  if (f_ct) {
    //log_e("Connection OK.");

    m_codec = CODEC_MP3;
    AACDecoder_FreeBuffers();
    if (!MP3Decoder_AllocateBuffers()) return false;
    sprintf(chbuf, "MP3Decoder has been initialized, free Heap: %u bytes", ESP.getFreeHeap());
    if (audio_info) audio_info(chbuf);
  }
  else {
    return false;
  }

  while (!playI2Sremains()) {
    ;
  }

  while (clientsecure.available() == 0) {
    ;
  }

  while (clientsecure.available()) {
    bytesCanBeWritten = InBuff.writeSpace();
    bytesAddedToBuffer = clientsecure.read(InBuff.getWritePtr(), bytesCanBeWritten);

    //contentLength += bytesAddedToBuffer;

    if (bytesAddedToBuffer > 0) InBuff.bytesWritten(bytesAddedToBuffer);
    bytesCanBeRead = InBuff.bufferFilled();
    if (bytesCanBeRead > InBuff.getMaxBlockSize()) bytesCanBeRead = InBuff.getMaxBlockSize();

    if (bytesCanBeRead) {
      while (InBuff.bufferFilled() >= InBuff.getMaxBlockSize()) { // mp3 frame complete?
        bytesDecoded = sendBytes(InBuff.getReadPtr(), InBuff.bufferFilled());
        InBuff.bytesWasRead(bytesDecoded);
        contentLength -= bytesDecoded;
      }
    }
  }
  while (InBuff.bufferFilled()) {
    bytesDecoded = sendBytes(InBuff.getReadPtr(), InBuff.bufferFilled());
    InBuff.bytesWasRead(bytesDecoded);
    contentLength -= bytesDecoded;
    if (contentLength < 100 || bytesDecoded < 10) break;
  }

  while (!playI2Sremains()) {
    ;
  }
  MP3Decoder_FreeBuffers();
  stopSong();
  clientsecure.stop();
  clientsecure.flush();
  m_codec = CODEC_NONE;
  //if (audio_eof_speech) audio_eof_speech(speech);
  return true;
}
</code>

nodoubtman, 2021-06-29:
I tried to add a function to play an mp3 from a website, but i received only this on the serial monitor: can you help?

info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 210744 bytes
info        SSL has been established, free Heap: 163328 bytes

and nothing more..

bool Audio::connecttositemp3(const char* host, const char* path) {
  setDefaults();
  bool     f_ct = false;
  uint32_t bytesCanBeWritten = 0;
  uint32_t bytesCanBeRead = 0;
  int32_t  bytesAddedToBuffer = 0;
  int16_t  bytesDecoded = 0;
  uint32_t contentLength = 0;

  String completePath = urlencode(path);

  String resp = String("GET ") + completePath + String(" HTTP/1.1\r\n")
                + String("Host: ") + host + String("\r\n")
                + String("User-Agent: ESP32 agent/1.0.0\r\n")
                + String("User-Agent: Mozilla/5.0 \r\n")
                + String("Accept-Encoding: identity\r\n")
                + String("Accept: text/html\r\n")
                + String("Connection: close\r\n\r\n");

  if (!clientsecure.connect(host, 443)) {
    log_e("Connection Failed!");
    return false;
  }

  log_e("Connection OK.");

  clientsecure.print(resp);
  sprintf(chbuf, "SSL has been established, free Heap: %u bytes", ESP.getFreeHeap());
  if (audio_info) audio_info(chbuf);
  while (clientsecure.connected()) { // read the header
    String line = clientsecure.readStringUntil('\n');
    if (line.startsWith("Content-Type")) {
      if (line.indexOf("audio/mpeg") > 12) f_ct = true; // Content-Type mpeg seen
    }
    line += "\n";
    if (line == "\r\n") break;
  }
  
  if (f_ct) {
    m_codec = CODEC_MP3;
    AACDecoder_FreeBuffers();
    if (!MP3Decoder_AllocateBuffers()) return false;
    sprintf(chbuf, "MP3Decoder has been initialized, free Heap: %u bytes", ESP.getFreeHeap());
    if (audio_info) audio_info(chbuf);
  }
  else {
    return false;
  }

  while (!playI2Sremains()) {
    ;
  }

  while (clientsecure.available() == 0) {
    ;
  }

  while (clientsecure.available()) {
    bytesCanBeWritten = InBuff.writeSpace();
    bytesAddedToBuffer = clientsecure.read(InBuff.getWritePtr(), bytesCanBeWritten);
    contentLength += bytesAddedToBuffer;
    if (bytesAddedToBuffer > 0) InBuff.bytesWritten(bytesAddedToBuffer);
    bytesCanBeRead = InBuff.bufferFilled();
    if (bytesCanBeRead > InBuff.getMaxBlockSize()) bytesCanBeRead = InBuff.getMaxBlockSize();
    if (bytesCanBeRead) {
      while (InBuff.bufferFilled() >= InBuff.getMaxBlockSize()) { // mp3 frame complete?
        bytesDecoded = sendBytes(InBuff.getReadPtr(), InBuff.bufferFilled());
        InBuff.bytesWasRead(bytesDecoded);
        contentLength -= bytesDecoded;
      }
    }
  }
  while (InBuff.bufferFilled()) {
    bytesDecoded = sendBytes(InBuff.getReadPtr(), InBuff.bufferFilled());
    InBuff.bytesWasRead(bytesDecoded);
    contentLength -= bytesDecoded;
    if (contentLength < 100 || bytesDecoded < 10) break;
  }

  while (!playI2Sremains()) {
    ;
  }

  MP3Decoder_FreeBuffers();
  stopSong();
  clientsecure.stop();
  clientsecure.flush();
  m_codec = CODEC_NONE;

  //if (audio_eof_speech) audio_eof_speech(speech);
  return true;
}

thankk you.
Marc.

schreibfaul1, 2021-06-29:
If the audio file has a valid ContentType and ContentLength on any server, it can be played, e.g. connecttohost('"https://cdn-storage.br.de/iLCpbHJGNL9zu6i6NL97bmWH_-4c/_-ZS/_AvH9Arf9U1S/210316_0000_Betthupferl_Dienstags-bei-Oma-Otto-25-Die-lustige-Gruse.mp3");

closed 2021-06-30 by nodoubtman