schreibfaul1/ESP32-audioI2S #173: Feature request: is it possible to make "m_buffSizeRAM" user-adjustable?

siy1, 2021-06-29:
First a big thanks for this excellent library. My small webradio project (ESP32 + PCM5102 + 16x2 I2C-LCD) worked out of the box. Output quality is good (connected to my old analog Yamaha stereo receiver), I have seen no drop-outs, even with high bandwidth (256k - 320k) mp3 or aac streams.

Now my possible feature request: the used buffer size is only 6399 bytes (according to the serial log - PSRAM is not available in my setup). Driven by curiosity I changed the value of the constant "m_buffSizeRAM" in src/Audio.h (default 1600 * 5) to higher values. Finally I was able to use even 1600 * 48 without memory problems which should give sufficient buffering  in the case of bad connections.
My question: Is it possible to make "m_buffSizeRAM" adjustable by the user, maybe controllable by an extra argument of Audio::Audio(...) or by a separate function like Audio::set_buffSizeRAM(...)? Or did I miss something and the feature is already available?

Thanks again & Kind regards!

schreibfaul1, 2021-07-02:
Hello siy1,
the audio buffer is set to the smallest usable size (~ about 5 mp3 frames). The available SRAM is scarce for large projects. If your project allows for an enlargement, you are welcome to do so, precisely because it can reduce the likelihood of dropouts.

siy1, 2021-07-05:
Dear schreibfaul1,

thanks for confirming that this is possible. In fact I already operate my small webradio with a buffer of 1600 * 48 without problems.

My question was more about making this changeable from the side of the user program without making modifications to your library files. A simple solution would be to replace the "1600 * 5" by a CPP macro (like BUFFSIZERAM), which could be defined by the user before calling "#include <Audio.h>". If not defined by the user a simple "#ifndef BUFFSIZERAM ..." could be used in Audio.h to set it to the default value of 1600 * 5.

In addition, I wonder why your library is not available via the Arduino library manager. I think this would help to make your nice work more visible - and would make updating more easy.

Thanks again!

closed 2021-09-19 by stale[bot]