schreibfaul1/ESP32-audioI2S #174: No sound with MAX98357A @ ESP32 WROOM 32U

brunnwart, 2021-06-29:
Hi all,

could someone give me an advice? I'm stuck with absolutely no sound via "standard" GPIO pins connected - I2S_DOUT  22 (or 25), I2S_BCLK  27 and I2S_LRC  26 - to the MAX98357A either reading the sd card (connected to 5, 23, 19 & 18) with
audio.connecttoFS(SD, "/mono.mp3");
or alternatively trying to stream using
audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.mp3"); 

Sound Hardware is OK, as playing from internal memory following the 2nd example (using the same pins but another lib) in https://diyi0t.com/i2s-sound-tutorial-for-esp32/ DOES work?!

Thanks a lot in advance!

trullock, 2021-08-23:
I've done exactly this and it works fine, youre going to have to post complete software and hardware details if you want help

brunnwart, 2021-08-28:
Thanks trullock for your response.
Well, the hw is an esp32dev, the Adafruit MAX98357A and a hw-203 sd card holder.
The code is the following (nothing special, most of it copy & paste from the examples):

#include "Audio.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
#define SD_CS          5    // grau
#define SPI_MOSI      23    // braun
#define SPI_MISO      19    // gelb
#define SPI_SCK       18    // lila
#define I2S_DOUT      22    // grau
#define I2S_BCLK      27    // gelb
#define I2S_LRC       26    // grün

Audio audio(true, I2S_DAC_CHANNEL_RIGHT_EN);

String ssid =     "ssid";
String password = "pw";

void logMemory() {
  log_d("Used PSRAM: %d", ESP.getPsramSize() - ESP.getFreePsram());
}

void setup(){
    pinMode(SD_CS, OUTPUT);
    digitalWrite(SD_CS, HIGH);
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    Serial.begin(115200);
    
    if (!SD.begin(SD_CS)) {
      Serial.println("SD Card initialization failed!");
      while (1);
    }
    uint8_t cardType = SD.cardType();

    if (cardType == CARD_NONE){
      Serial.println("No SD card attached");
      return;
    }

    Serial.print("SD Card Type: ");
    if(cardType == CARD_MMC){
      Serial.println("MMC");
    } else if(cardType == CARD_SD){
      Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
      Serial.println("SDHC");
    } else {
      Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);

    listDir(SD, "/", 0);

    WiFi.disconnect();
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid.c_str(), password.c_str());
    while (WiFi.status() != WL_CONNECTED) delay(1500);

    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21); // 0...21
    //audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.mp3"); //  128k mp3
    audio.connecttoFS(SD, "/mono.mp3");
}

void loop(){
  audio.loop();
  delay(60000);
}

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
  Serial.printf("Listing directory: %s\n", dirname);

  File root = fs.open(dirname);
  if(!root){
    Serial.println("Failed to open directory");
    return;
  }
  if(!root.isDirectory()){
    Serial.println("Not a directory");
    return;
  }

  File file = root.openNextFile();
  while(file){
    if(file.isDirectory()){
      Serial.print("  DIR : ");
      Serial.println(file.name());
      if(levels){
        listDir(fs, file.name(), levels -1);
      }
    } else {
      Serial.print("  FILE: ");
      Serial.print(file.name());
      Serial.print("  SIZE: ");
      Serial.println(file.size());
    }
    file = root.openNextFile();
  }
}

and the serial output is:
E (445) psram: PSRAM ID read error: 0xffffffff
SD Card Type: SDHC
SD Card Size: 7596MB
Listing directory: /
  FILE: /mono.mp3  SIZE: 205105

then it hangs there forever and nothing can be heard.
Thanks for any ideas!

schreibfaul1, 2021-08-28:
```
void loop(){
    audio.loop();
    delay(60000);
}
```
delay(60000)  its unneccessary

brunnwart, 2021-08-28:
> delay(60000) is unneccessary
O.K., thanks - I just tried different delays with no effect.

schreibfaul1, 2021-08-28:
The internal DAC is hard-wired. GPIO25 is channel 1 and GPIO26 is channel 2. SetPinout is not necessary

brunnwart, 2021-08-29:
> The internal DAC is hard-wired. GPIO25 is channel 1 and GPIO26 is channel 2. SetPinout is not necessary
O.K., but using the pinout 
#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26
for only one MAX98357A (mono) is correct, right?

schreibfaul1, 2021-08-29:
But you can either use the internal DAC or the external one, both of which do not work together.
If you want to use the MAX98357A change `Audio audio(true, I2S_DAC_CHANNEL_RIGHT_EN);` to `Audio audio;` without any further parameters. For only one channel you can switch to mono with `forceMono(true);`

brunnwart, 2021-08-29:
> If you want to use the MAX98357A change `Audio audio(true, I2S_DAC_CHANNEL_RIGHT_EN);` to `Audio audio;` without any further parameters. For only one channel you can switch to mono with `forceMono(true);`
O.K., that was it: using
Audio audio(true, I2S_DAC_CHANNEL_RIGHT_EN);
leads to no sound whereas 
Audio audio;
plays the mp3 file.
So this is solved, thanks a lot for your hint!
Now, scratches and other background sounds are as loud as the mp3 file to play - is there a possibility to reduce those disturbing sounds?

schreibfaul1, 2021-08-29:
Well, that's unusual, I suspect a bug in the wiring of the MAX. Perhaps BCLK has been swapped with LRC