schreibfaul1/ESP32-audioI2S #176: Uneble to listen stations with url extension .m3u8

ReyWamba, 2021-07-13:
Hello.

In my program, everything works fine with stations of all kinds, but I can't listen to those with the .m3u8 extension.

For example: [https://livefastly-webs.ondacero.es/ondacero/audio/master.m3u8](url) works both in browsers (with the corresponding extension) and on the iphone. However, I can't hear it on my esp32.

Any suggestion?

Thank you.

schreibfaul1, 2021-07-17:
Playlists in m3u8 format are not supported. Often these include audiovisual streams (e.g. IPTV) but in your example there is only one audio stream. The audio stream, however, does not run continuously, but is made up of many individual streams, see RFC8216 HTTP Live Streaming Pos. 4.3.2.1 EXTINF, https://datatracker.ietf.org/doc/html/rfc8216#section-4.2.

If you want to add that, the procedure is something like this:
https://livefastly-webs.ondacero.es/ondacero/audio/master.m3u8  unpacked
```
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-STREAM-INF:BANDWIDTH=131083,CODECS="mp4a.40.2"
chunklist.m3u8
```
then the next url is:
https://livefastly-webs.ondacero.es/ondacero/audio/chunklist.m3u8
unpacked:
```
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-TARGETDURATION:11
#EXT-X-MEDIA-SEQUENCE:367478
#EXTINF:10.048,
media-uvjuzfsj4_367478.aac
#EXTINF:9.984,
media-uvjuzfsj4_367479.aac
#EXTINF:9.984,
media-uvjuzfsj4_367480.aac
```
367478 ist the sequence number
call https://livefastly-webs.ondacero.es/ondacero/audio/media-uvjuzfsj4_367478.aac
it will play 10.048 seconds. 
in the log it looks like this:
```
audio_info: HTTP/1.1 200 OK
audio_info: Content-Length: 163260
audio_info: access-control-expose-headers: Date, Server, Content-Type, Content-Length
audio_info: access-control-allow-credentials: true
audio_info: access-control-allow-methods: OPTIONS, GET, POST, HEAD
audio_info: access-control-allow-headers: Content-Type, User-Agent, If-Modified-Since, Cache-Control, Range
audio_info: content-type: audio/x-aac, format is aac
audio_info: AACDecoder has been initialized, free Heap: 132864 bytes
audio_info: via: 1.1 varnish, 1.1 varnish, 1.1 varnish
audio_info: access-control-allow-origin: *
audio_info: date: Sat, 17 Jul 2021 17:06:18 GMT
audio_info: age: 25
audio_info: x-served-by: cache-mad22040-MAD, cache-fra19178-FRA, cache-fra19158-FRA
audio_info: x-cache: HIT, HIT, MISS
audio_info: x-cache-hits: 1, 1, 0
audio_info: x-timer: S1626541579.879782,VS0,VE1
audio_info: x-country: DE 
audio_info: x-geoblocking: yes
audio_info: Switch to SWM, bitrate is 0, metaint is 0
lasthost  : https://livefastly-webs.ondacero.es/ondacero/audio/media-uvjuzfsj4_369806.aac
audio_info: inputbuffer is being filled
audio_info: stream ready
audio_info: buffer filled in 16 ms
audio_info: syncword found at pos 73
audio_info: syncword found at pos 0
audio_info: Channels: 2
audio_info: SampleRate: 48000
audio_info: BitsPerSample: 16
audio_info: BitRate: 154875
audio_info: AAC HeaderFormat: ADTS
audio_info: AAC Codec: MPEG-4
audio_info: AAC Profile: LowComplexity
```
Content-Length is 163260
after163260 bytes are downloaded call 
call https://livefastly-webs.ondacero.es/ondacero/audio/media-uvjuzfsj4_367478.aac
etc.....

ReyWamba, 2021-08-30:
Hello again:

I replied a few days ago directly from the mail, but I see that it does not appear published.

I am not an advanced user programming ESP32 or in C ++, so I think adding the code to support the .m3u8 files is beyond my knowledge.

As I already mentioned, I have observed that several stations (not only in Spain, but even the BBC and others) are migrating from sound file formats (.mp3 or .acc) to list formats (such as .m3u8), leaving them obsolete many radios.

I have some doubts:

- How do I check that all the bytes of a file in the list have been downloaded to proceed to the next one?

- What happens when all the files in a list have finished downloading? Do I have to download the first .m3u8 file again to examine its contents?

- Do you plan to support these files in the future?

Thank you.

schreibfaul1, 2021-08-30:
This should work with the ESP-ADF IDE. I don't know if that's the case. To test that and find out how it is done is a lot of work. I don't have a board that is supported by the ADF. So I can't say whether this can be implemented in the near future.