schreibfaul1/ESP32-audioI2S #177: Feature Request - Callback with decoded data

craiglindley, 2021-07-26:
I'm trying to use your library to decode audio files but I want to stream them to a bluetooth speaker instead of using I2S for output. Having a callback available for decoded data would mean that any sink of the decoded data could be implemented without having to dive into and change the library's code.

Thanks for this library. It is the only one I have found that can decode m4a files.

schreibfaul1, 2021-07-27:
I already had the idea of ​​using Bluetooth instead of the DAC. As far as I know, BT works with a fixed sample rate of 48KHz. The sources usually have 44.1KHz. That doesn't really fit. For your own experiments, you will find a pointer to m_outBuff in the sendBytes() routine and the number of decoded bytes in validSamples.

craiglindley, 2021-07-27:
I've used ESP8266Audio successfully with BT with 44.1KHz CD quality audio
and I would use them again for my new project except they cannot decode m4a
files. They use the mp3_decoder as well.

If you built in a callback for decoded data you could default it to I2S
output and everything would be the same.

Just a thought.

-- 
Craig Lindley / Heather Hubbard

New Recordings are here <http://craigandheather.net/spellbound.html>

Personal Website is here <http://craigandheather.net>

Please call the Rockrimmon house. Our cell phones don't work at home.
Rockrimmon House: (719) 426-9873
Craig's Cell: (719) 502-7925
Heather's Cell: (719) 571-0944

If you’re one in a million, there are now more than seven thousand people
exactly like you.

closed 2021-12-09 by stale[bot]