schreibfaul1/ESP32-audioI2S #178: FLAC format without sound

SyXavier, 2021-07-28:
ESP32 WROVER IE and PCM5102A play FLAC format without sound, but MP3 format is normal. The log is as follows:
file name:/yi lu xiang bei.flac
info        PSRAM found, inputBufferSize: 283615 bytes
info        buffers freed, free Heap: 203152 bytes
info        Reading file: "/yi lu xiang bei.flac"
info        FLACDecoder has been initialized, free Heap: 202828 bytes
info        stream ready
info        Content-Length: 35528039

schreibfaul1, 2021-07-28:
There are many flac header versions.
If the FLAC frame is> 16K, the file will not be played due to lack of resources. But then there is a message. Can you send me the file?

SyXavier, 2021-07-29:
Okay, I sent the file to your email.

closed 2021-12-09 by stale[bot]