schreibfaul1/ESP32-audioI2S #179: Can't Compile with PlatformIO

Cal-237, 2021-08-02:
I wasn't sure where to put this so I apologize if "issues" is the wrong place. I am using platformIO with the Arduino framework to program my esp32. I added your library to the source files, but I get a lot of errors when I try to compile.

Here are the errors:

.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio15read_OGG_HeaderEPhj+0x28): undefined reference to `FLACDecoder_AllocateBuffers()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio16parseContentTypeEPKc+0x54): undefined reference to `MP3Decoder_AllocateBuffers()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio16parseContentTypeEPKc+0x58): undefined reference to `AACDecoder_AllocateBuffers()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio15showCodecParamsEv+0x2c): undefined reference to `AACGetFormat()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio15showCodecParamsEv+0x30): undefined reference to `AACGetID()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio15showCodecParamsEv+0x34): undefined reference to `AACGetProfile()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio12findNextSyncEPhj+0x18): undefined reference to `MP3FindSyncWord(unsigned char*, int)'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio12findNextSyncEPhj+0x1c): undefined reference to `AACFindSyncWord(unsigned char*, int)'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio12findNextSyncEPhj+0x20): undefined reference to `AACSetRawBlockParams(int, 
int, int, int)'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio12findNextSyncEPhj+0x24): undefined reference to `FLACSetRawBlockParams(unsigned char, unsigned int, unsigned char, unsigned int, unsigned int)'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio12findNextSyncEPhj+0x28): undefined reference to `FLACFindSyncWord(unsigned 
char*, int)'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio24compute_audioCurrentTimeEi+0x1c): undefined reference to `MP3GetBitrate()'.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio24compute_audioCurrentTimeEi+0x20): undefined reference to `AACGetBitrate()'.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio24compute_audioCurrentTimeEi+0x24): undefined reference to `FLACGetBitRate()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio15read_M4A_HeaderEPhj$part$12$constprop$17+0x98): undefined reference to `MP3Decoder_ClearBuffer()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio15read_M4A_HeaderEPhj$part$12$constprop$17+0x9c): undefined reference to `FLACDecoderReset()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x14): undefined reference to `MP3Decode(unsigned char*, int*, short*, int)'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x18): undefined reference to `AACDecode(unsigned char*, int*, short*)'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x1c): undefined reference to `FLACDecode(unsigned char*, int*, short*)'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x20): undefined reference to `MP3GetChannels()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x24): undefined reference to `MP3GetSampRate()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x28): undefined reference to `MP3GetBitsPerSample()'        
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x2c): undefined reference to `AACGetChannels()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x30): undefined reference to `AACGetSampRate()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x34): undefined reference to `AACGetBitsPerSample()'        
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x38): undefined reference to `FLACGetChannels()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x3c): undefined reference to `FLACGetSampRate()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x40): undefined reference to `FLACGetBitsPerSample()'       
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x44): undefined reference to `MP3GetOutputSamps()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x48): undefined reference to `AACGetOutputSamps()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio9sendBytesEPhj+0x4c): undefined reference to `FLACGetOutputSamps()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio16processLocalFileEv+0x14): undefined reference to `MP3Decoder_FreeBuffers()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio16processLocalFileEv+0x18): undefined reference to `AACDecoder_FreeBuffers()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o):(.literal._ZN5Audio16processLocalFileEv+0x1c): undefined reference to `FLACDecoder_FreeBuffers()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o): In function `Audio::read_OGG_Header(unsigned char*, unsigned int)':
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `FLACDecoder_AllocateBuffers()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o): In function `Audio::parseContentType(char const*)':
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `MP3Decoder_AllocateBuffers()'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `AACDecoder_AllocateBuffers()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o): In function `AudioBuffer::changeMaxBlockSize(unsigned short)':
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `MP3Decoder_AllocateBuffers()'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `AACDecoder_AllocateBuffers()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o): In function `Audio::parseContentType(char const*)':
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `FLACDecoder_AllocateBuffers()'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `FLACDecoder_AllocateBuffers()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o): In function `Audio::showCodecParams()':
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `AACGetFormat()'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `AACGetID()'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `AACGetProfile()'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `AACGetID()'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `AACGetProfile()'
.pio\build\nodemcu-32s\libc31\libESP32-audioI2S-master.a(Audio.cpp.o): In function `Audio::findNextSync(unsigned char*, unsigned int)':
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `MP3FindSyncWord(unsigned char*, int)'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `AACFindSyncWord(unsigned char*, int)'
[... cut ...]
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `MP3Decoder_AllocateBuffers()'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `AACDecoder_AllocateBuffers()'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `AACDecoder_AllocateBuffers()'
C:\Projects\ESP32\music_test/lib\ESP32-audioI2S-master/Audio.cpp:3527: undefined reference to `FLACDecoder_AllocateBuffers()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nodemcu-32s\firmware.elf] Error 1

schreibfaul1, 2021-08-02:
The files do not have to be copied by hand, PlatformIO does that itself.
The platformio.ini could look like this:
````
 PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/en/latest/platforms/espressif32.html

[env:esp32dev]
platform = espressif32
board = esp32dev
board_build.f_cpu = 240000000L
framework = arduino
monitor_speed = 115200
monitor_filters = direct, esp32_exception_decoder
monitor_flags = --raw
board_build.partitions = huge_app.csv

lib_deps =
    https://github.com/schreibfaul1/ESP32-audioI2S

platform_packages =
;    platformio/framework-arduinoespressif32_105 @ https://github.com/espressif/arduino-esp32.git#1.0.5
;    platformio/framework-arduinoespressif32_106 @ https://github.com/espressif/arduino-esp32.git#1.0.6
;    framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.0-alpha1

    platformio/tool-esptoolpy @ ~1.30100
;    toolchain-xtensa32 @ ~2.80400.0
    platformio/framework-arduinoespressif32_106 @ https://github.com/espressif/arduino-esp32.git#1.0.6

board_upload.maximum_size = 3145728
board_upload.flash_size = 4MB

;build_flags = -DCORE_DEBUG_LEVEL=0    ; None
;build_flags = -DCORE_DEBUG_LEVEL=1    ; Error
;build_flags = -DCORE_DEBUG_LEVEL=2    ; Warn
;build_flags = -DCORE_DEBUG_LEVEL=4    ; Debug
;build_flags = -DCORE_DEBUG_LEVEL=5    ; Verbose
build_flags = 
        -DCORE_DEBUG_LEVEL=5            ; Info
        -DCONFIG_ARDUHAL_LOG_COLORS=1
;        -DBOARD_HAS_PSRAM
;        -mfix-esp32-psram-cache-issue
````

Cal-237, 2021-08-02:
Thank you so much! Now it compiles, however it does not seem to work with the DAC.

My entire code is here (but of course the ssid and password are replaced for the correct ones):

```
#include <Arduino.h>
#include "WiFi.h"
#include "Audio.h"
#include "SPIFFS.h"

Audio audio(true, I2S_DAC_CHANNEL_LEFT_EN);

void setup() {
    Serial.begin(9600);
    WiFi.disconnect();
    WiFi.mode(WIFI_STA);
    WiFi.begin("*******", "********");
    while (WiFi.status() != WL_CONNECTED) delay(500);
    Serial.println("WiFi connected");
    audio.setVolume(21); // 0...21
    audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.mp3");
}

void loop() {
  audio.loop();
}
```

I have earbuds connected to GPIO pin 26 through a 220 Ohm resistor, and the ground side connected to ground. I am getting the WiFi connected message from the Serial Monitor, but I only hear a loud flickering noise through the earbud and then silence.

schreibfaul1, 2021-08-02:
du kannst vorab die I2S Pins festlegen 

#define I2S_SDOUT   25
#define I2S_BCLK      27
#define I2S_LRCK      26

Audio audio;

void setup(){
    ...
    audio.setPinout(I2S_BCLK, I2S_LRCK, I2S_SDOUT);
    audio.setVolume(21); // 0...21
    audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.mp3");
    ....

Cal-237, 2021-08-02:
I don't want to use i2s, I just want to use the built-in DAC on pin 26 to play music. That is why I put `(true, I2S_DAC_CHANNEL_LEFT_EN)` in the constructor for the audio object.

closed 2021-12-09 by stale[bot]

Protonerd, 2022-11-17:
Dear schreibfaul1, even though I'm trying to compile for Arduino/C++ , I have similar issues, so I hope it's OK that I did not open a new issue, but rather pick up the thread here. What am I doing wrong? The files under src are there.

C:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\libraries\ESP32-audioI2S-master\src/Audio.cpp:4434: undefined reference to `FLACDecoderReset()'
C:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\libraries\ESP32-audioI2S-master\src/Audio.cpp:4434: undefined reference to `MP3Decoder_FreeBuffers()'
C:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\libraries\ESP32-audioI2S-master\src/Audio.cpp:4434: undefined reference to `AACDecoder_FreeBuffers()'
C:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\libraries\ESP32-audioI2S-master\src/Audio.cpp:4434: undefined reference to `AACDecoder_FreeBuffers()'
C:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\libraries\ESP32-audioI2S-master\src/Audio.cpp:4434: undefined reference to `FLACDecoder_FreeBuffers()'

schreibfaul1, 2022-11-17:
I know Sloeber and like to use it. You won't have made a mistake, everything looks fine. Sometimes I also had such error messages without a recognizable reason. It often helps to delete the 'Release' folder in the project and create everything again with 'built all'.

Protonerd, 2022-11-18:
Hallo Wolle,

Leider besteht das Problem hartnackig weiter, obwohl ich mehrmals das Release Ordner geloscht habe mit einem anschliessenden Clean. Build funktioniert, die Fehlermeldung passiert erst wenn der combiner gestartet wird.
Ich benutze Sloeber 4.3 (also nicht das Neueste) aber ich denke nicht, dass das Problem dadurch verursacht ware.

'Starting combiner'
"C:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:\Users\kunandra\Arduino\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib" "-LC:\Users\kunandra\Arduino\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.6/tools/sdk/ld" -T esp32_out.ld -T esp32.project.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u esp_app_desc -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u __cxx_fatal_exception   -Wl,--start-group    .\sloeber.ino.cpp.o  .\libraries\arduino-audio-tools-main\main.cpp.o  .\libraries\Wire\src\Wire.cpp.o  .\libraries\WiFi\src\ETH.cpp.o

schreibfaul1, 2022-11-18:
Schade, jetzt fällt mir auch nicht mehr sehr viel ein. Die Arduino Version 1.0.6 zu verwenden ist in Ordnung. Aber die Fehlermeldung ist jetzt anders. Hast du schon mal versucht die V1.0.6 zu entfernen nund neu zu holen? In der IDE unter Window->Preferences->Sloeber->Platforn and Boards->ESP32 den Haken entfernen. Dann wird der Ordner C:\Users\kunandra\Arduino\Sloeber/arduinoPlugin/packages/esp32/hardware/esp32/1.0.6 gelöscht. Und wenn du den Haken setzt wird V1.0.6 neu installiert. 
Wenn das auch nicht hilft probiere, ob nicht ein kleines 20 Zeilen Demoprogr. von meinem GithubRepo funktioniert.

Protonerd, 2022-11-21:
Hallo Wolle, ich habe jetzt Sloeber neu installiert und alles aufgeraumt. Das Beispielcode I2Saudio.io funktioniert jetzt, allerdings nur wenn ich Internetradio uber WiFi als Quelle definiere. Lesen und Abspielen von SD-Karte geht aber leider nicht, obwohl sowohl Verbindung zum MAX98357 als auch die Verdrahtung von der SD-Karten-Breakout richtig sind. Letzteres habe ich mit einen Print der Verzeichnisse uberpruft:
```
 rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
/System Volume Information
/SF4
/SF5
/CfgSnd
/music
/SF1
/SF2
/SF3
/test.wav
info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 196472 bytes
info        Reading file: "/test.wav"
```
als test file habe ich - nachdem meine Files nicht abgespielt werden konnten - das Pink-Panther File von deinem Repo genommen und umbenannt.
Was konnte das Problem sein ? (wenn du magst kann ich eine neue Issue aufmachen, weil es thematisch nicht zu dieser Issue passt)

schreibfaul1, 2022-11-21:
Das sieht soweit richtig gut aus, die SD Karte funktioniert und das Verzeichnis wird ausgegeben. Ich sehe, dass du ohne ohne PSRAM arbeitest. Ohne PSRAM mit wav gibt es ein Problem, ist mir nicht aufgefallen, da ich meist PSRAM benutze. Lade bitte die Version, die ich jetzt eingestellt habe herunter, dann sollte das klappen.
mein Log:
````
info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 162432 bytes
info        Reading file: "/wave_test/Pink-Panther.wav"
info        FormatCode: 1
info        DataRate: 176400
info        DataBlockSize: 4
info        BitsPerSample: 16
info        Audio-Length: 1251992
info        stream ready
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: 1411200
info        Closing audio file
info        End of file "Pink-Panther.wav"
eof_mp3     Pink-Panther.wav
````

Protonerd, 2022-11-22:
Noch mal was gelernt, ich wusste vorher nicht, dass man eine Speichererweiterung hat. Leider haben beide meiner ESP32 Boards (Adafruit Huzzah32 und ESP32 DevKitV1) kein PSRAM... Ich habe jetzt dein Update aufgespielt, wav File fängt jetzt an abgespielt zu werden, aber im Gegensatz zum WiFi Radio höre ich nur Geräusche. MP3-Beispielfile 320k_test.mp3 verursacht folgende Fehlermeldung (s.u.). Ich möchte besser verstehen, warum man PSRAM benötigt, es liegt zwar auf der Hand, dass man damit großere Datenmengen einlesen kann, bei einem anderen uC habe ich für Wav-Abspielen 2 Buffers von jeweils 512bytes benutzt und während der eine gespielt wurde, wurde der andere gefüllt. Hat das ESP32 auch DMA für I2S? Im Prinzip was ich mit dem ESP32 gerne machen möchte ist mehrere (1-5) Wav-Files einzulesen und mischen (sog. Polyphinic playback)
```
info        buffers freed, free Heap: 196212 bytes
info        Reading file: "/320k_test.mp3"
info        MP3Decoder has been initialized, free Heap: 172588 bytes
info        Content-Length: 9610227
info        file has no mp3 tag, skip metadata
info        Audio-Length: 9610227
info        stream ready
info        syncword not found
[I][Audio.cpp:3402] playAudioData(): err bytesDecoded -1
[I][Audio.cpp:3402] playAudioData(): err bytesDecoded -1
```