schreibfaul1/ESP32-audioI2S #190: Can't use other languages

SyXavier, 2021-08-18:
Hi, Wolle. Can't sdfat example use other languages?
----------------------------------------------------------------------------------------------------
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10864
load:0x40080400,len:6432
entry 0x400806b8
info        PSRAM found, inputBufferSize: 283615 bytes
info        buffers freed, free Heap: 312380 bytes
info        Reading file: "/蜀绣.mp3"
info        Failed to open file for reading
----------------------------------------------------------------------------------------------------
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10864
load:0x40080400,len:6432
entry 0x400806b8
info        PSRAM found, inputBufferSize: 283615 bytes
info        buffers freed, free Heap: 312380 bytes
info        Reading file: "/良い一日私の友達.mp3"
info        Failed to open file for reading
----------------------------------------------------------------------------------------------------

trullock, 2021-08-23:
Does `sd.h` support such file names?

SyXavier, 2021-08-23:
Doesn't SdFat.h support UTF8 encoded file name?

trullock, 2021-08-23:
Thats what I was asking and suggesting you test. FAT32 does support them, but I'm not sure (and tbh i doubt) Arduino does. Maybe I'm wrong...

trullock, 2021-08-23:
If you look in the source, theres `SDFATFS_USED` which you need to define to support UTF-8 filenames

SyXavier, 2021-08-23:
I have defined SDFATFS_USED, but it didn't work.

schreibfaul1, 2021-08-23:
It is possible that SDFAT V2 does not support these 蜀绣 characters.
https://github.com/greiman/SdFat/issues/200

SyXavier, 2021-09-25:
Hi, Wolle. I tested the SdFat library and it does support Unicode encoding. But using SdFat in the audio library does not open the file.

closed 2022-01-09 by stale[bot]