schreibfaul1/ESP32-audioI2S #193: Format .pls not working

tv4you2016, 2021-08-24:
hello @schreibfaul1 

I verified that your lib is compatible with the .pls format

but this is not 100% functional.

I have this url:
http://centova.radios.pt:8401/;listen.pls
and it doesn't work.

here I put a small log

PSRAM found, inputBufferSize: 1983615 bytes
[ 12054][V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
info
freed buffers, free Heap: 140395 bytes
info
Connect to new host: "http://centova.radios.pt:8401/;listen.pls"
info
Connect to "centova.radios.pt" on port 8401, extension "/;listen.pls"
info
Connected to server in 25 ms
info
read from playlist
info
Playlistheader: HTTP/1.0 200 OK
info
Playlistheader: icy-notice1:<BR>This stream requires <a href="http://www.winamp.com">Winamp</a><BR>
info
Playlistheader: icy-notice2:Shoutcast DNAS/posix(linux x64) v2.6.0.753<BR>
info
Playlistheader: Accept-Ranges:none
info
Playlistheader: Access-Control-Allow-Origin:*
info
Playlistheader: Cache-Control:no-cache,no-store,must-revalidate,max-age=0
info
Playlistheader: Connection:close
info
Playlistheader: icy-name:Radio ORBITAL
info
Playlistheader: icy-genre:Dance
info
Playlistheader: icy-br:128
info
Playlistheader: icy-sr:44100
info
Playlistheader: icy-url:http://www.orbital.pt
info
Playlistheader: icy-pub:0
info
Playlistheader: content-type:audio/mpeg
info
Playlistheader: icy-metaint:8192
info
Playlistheader: X-Clacks-Overhead:GNU Terry Pratchett
info
Switch to PLAYLISTDATA
info
Playlistdata:
info
processPlayListData== FORMAT_PLS
info
Playlistdata: `>[BP]+{$#msG`~/( @#E2=:YZ[l^5DLuXZa/s44SW7_X{*,@ZCePWBDK"0Vya;+OB
info

schreibfaul1, 2021-08-24:
Hello tv4you2016,
thanks for the message, this is a very rare case. The URL does not point to a playlist, but starts immediately with the audio header.
Radio Orbital is now working.

closed 2021-08-24 by schreibfaul1

tv4you2016, 2021-10-04:
Hello. this fix was in which lib ?? and it still doesn't work, apparently only src from ESP32-vs1053_ext and ESP32-audioI2S still doesn't work

nodoubtman, 2022-08-07:
> Hello. this fix was in which lib ?? and it still doesn't work, apparently only src from ESP32-vs1053_ext and ESP32-audioI2S still doesn't work

@schreibfaul1 you have modified VS1053 library not ESP32-I2S :(

can you do it?

Thanks.
Marc.

schreibfaul1, 2022-08-08:
Hello Marc,
I'm experimenting with m3u8 streams. These have more extensive playlists and therefore I have changed some of the processing. There are several ways in which the server delivers the playlists, with a value for contentlength, without contentlength but defined as chunked, or without anything. I corrected a few things and I hope everything works again. If the error still persists, please send me a broken url.

commit 31ace77f (2021-08-24): fix  #193   Format .pls not working