schreibfaul1/ESP32-audioI2S #211: Most streams fail to connect or require multiple attempts.

mikerussellnz, 2021-10-14:
Most streams including the example fail to connect.  About 1 in 20 connects and sometimes requires multiple attempts.  When it does connect, the stream plays fine.

I have checked Wifi is connected and putting a url like http://www.google.com in the serial console shows the HTTP headers so connections can be made.  

Example of a failure:
info        Connect to new host: "http://mp3.ffh.de/radioffh/hqlivestream.mp3"
info        Connect to "mp3.ffh.de" on port 80, extension "/radioffh/hqlivestream.mp3"
info        Request http://mp3.ffh.de/radioffh/hqlivestream.mp3 failed!
station     
streamtitle 
icyurl      

Is there any way to turn up the debug logging to find out why the connection is failing?

mikerussellnz, 2021-10-14:
Turning up the ESP logging I see that the connect timeout is 250ms and times out but I don't know why, perhaps the timeout is too low?

info        buffers freed, free Heap: 225392 bytes
info        Connect to new host: "http://198.15.94.34:8006/stream"
info        Connect to "198.15.94.34" on port 8006, extension "/stream"
[I][WiFiClient.cpp:243] connect(): select returned due to timeout 250 ms for fd 55
info        Request http://198.15.94.34:8006/stream failed!
station     
streamtitle 
icyurl

mikerussellnz, 2021-10-14:
After increasing the timeout to > 1000ms it connects, but gets slow stream and dropouts.  It seems something is wrong between the ESP32 and my Wifi AP?

info        Connect to new host: "http://mp3.ffh.de/radioffh/hqlivestream.mp3"
info        Connect to "mp3.ffh.de" on port 80, extension "/radioffh/hqlivestream.mp3"
info        Connected to server in 990 ms
info        HTTP/1.0 200 OK
bitrate     128000
info        icy-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
icyurl      https://www.ffh.de
info        instance-id: d93668d93bcc3c3ced9edeb191dda88f
info        server: AIS Streaming Server 8.4.3
info        icy-name: HIT RADIO FFH
...

schreibfaul1, 2021-10-14:
Thanks for pointing out that a compromise has to be found here between timeout and waiting time that is too long for destinations that cannot be reached.

In my case
audio_info   : Connect to new host: "http://mp3.ffh.de/radioffh/hqlivestream.mp3"
audio_info   : Connected to server in 77 ms

That certainly depends on how fast the Internet is, but increasing the timeout time may be a good solution

closed 2022-03-02 by stale[bot]

ideaalab, 2023-06-05:
Hey, sorry to open this again. But is there a define I can use to increase the timeout? How can I change it?
Thanks