schreibfaul1/ESP32-audioI2S #217: DC shift in MP3 decoding

rimim, 2021-10-30:
I've been tracking the source of clicks between my sound files and it seems that the MP3 decoder is introducing a software DC offset. If I encode a one second silence wav (signed 16-bit mono) file into an MP3 and then decode that MP3 on the ESP32 the bytes being sent via sendBytes() are all 0x2000 instead of 0x0000. This offset seems constant as everything is shifted up. This offset is not seen if I decode using LAME instead.

Here's an original example file:
[image]

Here's the decoded example output:
[image]

I'll continue to investigate, but thought I'd share my findings so far.

schreibfaul1, 2021-10-31:
That's an interesting observation. In the playSample (int16_t sample [2]) routine, I used to output the samples, precisely at the point where I2S is written.
````c++
    static int i = 0;
    i ++;
    if (i == 1000) {
        i = 0;
        log_i ('right% i', (s32 >> 8) & 0xffff);
        log_i ('left% i', s32 & 0xffff);
    }

    esp_err_t err = i2s_write ((i2s_port_t) m_i2s_num, (const char *) & s32, sizeof (uint32_t), & m_i2s_bytesWritten, 1000);
````   
Only every 1000 samples, otherwise too much data will be created. I have at the beginning of a song where there is still no offset.

````
audio_info   : BitsPerSample: 16
audio_info   : BitRate: 320000
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 0
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 0
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 0
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 0
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 0
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 0
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 0
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 0
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 0
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 0
[I][Audio.cpp:4165] playSample(): left 0
[I][Audio.cpp:4166] playSample(): right 2
[I][Audio.cpp:4165] playSample(): left 255
[I][Audio.cpp:4166] playSample(): right 65533
[I][Audio.cpp:4165] playSample(): left 255
[I][Audio.cpp:4166] playSample(): right 65532
[I][Audio.cpp:4165] playSample(): left 255 
````

rimim, 2021-10-31:
While investigating this I learned that the MP3 format produces a silence delay at the beginning and silence padding at the end (for historic reasons probably having to do with early hardware). If you encode a WAV file to MP3 and decode it you will end up with a longer length WAV file. The LAME encoder encodes the number of delay and padding samples into the beginning frame header so you can skip the silence, but this is an extension and not part of the MP3 specification. Lots of software uses LAME or otherwise supports this extension. So if you try this on the desktop you will most likely get the same length file back. You would have to write your own small MP3 encoder using the LAME library and tell it not to write this extension header to reproduce.

The libhelix MP3 decoder is older and does not support this LAME extension, so it will output silence in the beginning and at the end. Once it starts outputting the actual audio samples the "DC" shift happens.

For testing I modified the library to just send the generated samples over UDP and record them using a small python script.