schreibfaul1/ESP32-audioI2S #218: New  board  A1S v2.2 3378

ZnakZorro, 2021-11-02:
I got the board ESP32 A1S Audio Kit, v2.2 3378
I2C is on the pins
#define IIC_CLK 32
#define IIC_DATA 33
but I don't know what codec and what pins I2S

schreibfaul1, 2021-11-02:
Please have a look in this forum, the contribution from amiga from July 5th, 2021
https://www.mikrocontroller.net/topic/474383#new

ZnakZorro, 2021-11-04:
This example does not work for me.
By trial and error, I found:
// I2S GPIOs
#define I2S_SDOUT     26
#define I2S_BCLK      27
#define I2S_LRCK      25
#define I2S_MCLK       0
#define I2S_DSIN      35
// I2C GPIOs
#define IIC_CLK       32
#define IIC_DATA      33
// Amplifier enable
#define GPIO_PA_EN    19

Wyatt59, 2021-11-10:
Hello ,
it's work for me with this (with an old relise)
#define I2S_DSIN      25
#define I2S_BCLK      27
#define I2S_LRC       26
#define I2S_MCLK       0
#define I2S_DOUT      35
#define GPIO_PA_EN    21

// I2C GPIOs
#define IIC_CLK                     32
#define IIC_DATA                    33

schreibfaul1, 2021-11-10:
This corresponds to the pin assignment of my board, I have the V2.2 2379. ZnakZorro has a newer one, I have adopted the pin definition for both boards in ESP32_A1S.ino

ZnakZorro, 2021-11-10:
But ESP32 A1S Audio Kit, v2.2 3378 working with ES8388 codec not AC101

Wyatt59, 2021-11-10:
ok I find this 
https://www.deeptronic.com/store/blackstomp-core/esp32-a1s-ai-thinker-audio-module/
if it's can help...

schreibfaul1, 2021-11-10:
I've already read that AI Thinker uses the ES8388 in the new A1S board. It wasn't clear to me that the V2.2 3378 is related to it. Thanks for the hint.

closed 2022-04-17 by stale[bot]