schreibfaul1/ESP32-audioI2S #225: Is it possible to play wav in hex format from progmem?

unregistered33, 2021-11-11:
wav file is presented as an array of int? Thanks!

schreibfaul1, 2021-11-11:
Not directly, a file format with the wav header ('RIFF ...') must be available. You can upload and play a wav file in SPIFFS

closed 2022-03-13 by unregistered33

AndreKR, 2023-09-15:
I think this issue was about the _source_ of the file (byte array instead of SD card), not about the _format_ (RIFF header or not).

I'm also wondering if it would be possible to play a short MP3 from a byte array or (even better because it would work in ESPHome without changes to the ESPHome code) from a data URI.

schreibfaul1, 2023-09-16:
I don't know what data format ESPHome requires. Basically, the wav must have a RIFF header. This identifies the size of the audio block, the sample rate, the number of channels and the number of bits per sample. You do not necessarily need an SD card. The audio file can also be saved as SPIFFS or FFat.
And you're not limited to wav. To reduce the file size you can use a compressed format like mp3 or opus.