schreibfaul1/ESP32-audioI2S #227: ESPuino, play audio at special position ->Watchdog reset (TG1WDT_SYS_RESET)

tueddy, 2021-11-11:
Hi Wolle,

i use the ESPuino play/pause feature, so audio starts at any position.
Now i get a watchdog reset, assume something hangs in a endless loop. Log:

info        : PSRAM found, inputBufferSize: 283615 bytes
info        : buffers freed, free Heap: 132852 bytes
info        : Reading file: "/kaeptnblaubaer/wdr2_2018-10-03_kaeptnblaubaerderkoenigderdiebe_wdr2.mp3"
info        : MP3Decoder has been initialized, free Heap: 129540 bytes
Titel wird abgespielt ab Position **_1418645_**
'/kaeptnblaubaer/wdr2_2018-10-03_kaeptnblaubaerderkoenigderdiebe_wdr2.mp3' wird abgespielt (1 von 2)
info        : stream ready
info        : Content-Length: 3160374
info        : file has no mp3 tag, skip metadata
info        : Audio-Length: 3160374
info        : syncword found at pos 17
info        : syncword found at pos 0
info        : MP3 decode error -3 : FREE_BITRATE_SYNC
info        : syncword found at pos 121
info        : syncword found at pos 0
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4 

To reproduce the bug attached the mp3, start audio at position _1418645_

Schönen Dank für Deine Mühe
Tueddy

[wdr2_2018-10-03_kaeptnblaubaerderkoenigderdiebe_wdr2.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/7523016/wdr2_2018-10-03_kaeptnblaubaerderkoenigderdiebe_wdr2.zip)

schreibfaul1, 2021-11-11:
Hello tueddy,
the error seems to be extremely rare. It shouldn't appear at pause / resume? But it is reproducible if you jump exactly to the specific position in the file.

Best regards
Wolle

closed 2021-11-11 by schreibfaul1

tueddy, 2021-11-12:
This commit fixed the crash!
Vielen Dank !!!