schreibfaul1/ESP32-audioI2S #229: How to correctly stop playback and run again?

unregistered33, 2021-11-17:
Now I have a structure as shown below, but from time to time when I try to start playback before the current one ends, my device reboots:
`
void preparePlay(const char *path, uint8_t vol)
{
    if (audio.isRunning())
        audio.stopSong();
    needPlay = false;
    audio.connecttoFS(SD, path);
    audio.setVolume(vol);
}

void playAudio()
{
    needPlay = true;
}

void loop()
{
    if (isEventOccurs) //from interrupt
    {
        preparePlay(door_bell_path, volume);
        playAudio();
    }
    if (needPlay)
    {
        audio.loop();
    }
}`
Thanks!

void audio_eof_mp3(const char* info)
{
   needPlay = false;
}

unregistered33, 2021-11-17:
I'm so sorry... The problem not with ESP32-audioI2S. Its worked perfect! I am using a library in conjunction with an asynchronous webserver and put audio.connecttoFS in a server callback. This was fundamentally wrong!

closed 2021-11-17 by unregistered33

schreibfaul1, 2021-11-17:
great, you did it yourself. My solution to interrupt a song for the doorbell would be:
````c++
#include "Arduino.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"
#include "Ticker.h"

// Digital I/O used
#define SD_CS          5
#define SPI_MOSI      23
#define SPI_MISO      19
#define SPI_SCK       18
#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26

Audio audio;
Ticker ticker;

String playlist[] = {
        "title1.mp3",
        "title2.mp3",
        "title3.mp3",
        "title4.mp3",
        //....
        //....
};
const uint8_t num_titles = sizeof(playlist)/sizeof(playlist[0]);
int i;
uint32_t playpos;
boolean f_bell = false;
boolean f_bell_is_active = false;

void bell(){ // simulates a ISR
    if(f_bell_is_active) return; // ignore this, because the bell is already ringing
    f_bell = true;
}

void setup() {
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    Serial.begin(115200);
    SD.begin(SD_CS);
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21); // 0...21
    ticker.attach(20, bell); // simulate doorbell all 20s
    i = 0;
    audio.connecttoFS(SD, playlist[i].c_str());
}

void loop()
{
    if(f_bell){
        playpos = audio.getFilePos();
        audio.connecttoFS(SD, "bell.mp3");
        f_bell_is_active = true;
        f_bell = false;
    }
    audio.loop();
}

void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}

void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);

    if((String)info == "bell.mp3"){
        f_bell_is_active = false;
        audio.connecttoSD(playlist[i].c_str());
        audio.setFilePos(playpos); // set old playpos, pauseresume will not work in this case
        return;
    }

    i++;
    if(i == num_titles) i = 0;
    audio.connecttoSD(playlist[i].c_str());
}

````

livebelive, 2022-01-01:
Thanks for the lib first!
I am get a problem 
It is my problem but I holp  you can help me thank you very much!
I try link the lib to esphome make it work for homeassisyant
Esphome creat on_play_media and on_stop_media service for homeassistant
when service call on_play then send a url to esp32 
Now I need get status for play
if start play then send ststus on to homeassistnt
esle if music stop send  ststus off

maybe you do not know esphome and homeassistant
but any answer is great!

livebelive, 2022-01-01:
The problem is that the device will crash and restart during use, or it will always be in an infinite loop under certain conditions