schreibfaul1/ESP32-audioI2S #231: Plop sound at end and begin of TTS

deladriere, 2021-11-25:
Here is my code. Maybe I am doing something wrong ?
There is a plop sound at the beginning and at the end 

```
// from https://github.com/schreibfaul1/ESP32-audioI2S
// simplified by https://www.xtronical.com/i2sinternetradio/
// ESP32 Dev Module : ok
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"

// Digital I/O used
#define I2S_DOUT      25  // DIN connection
#define I2S_BCLK      27  // Bit clock
#define I2S_LRC       26  // Left Right Clock

Audio audio;
bool eos=1;

String ssid =     "#####"; // h 
String password = "#####"; //

void setup() {
    Serial.begin(115200);
    pinMode(22,INPUT_PULLUP);
    WiFi.disconnect();
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid.c_str(), password.c_str());
    while (WiFi.status() != WL_CONNECTED) delay(1500);
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(15); // 0...21

//    audio.connecttohost("http://www.wdr.de/wdrlive/media/einslive.m3u");
//    audio.connecttohost("http://macslons-irish-pub-radio.com/media.asx");
//    audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.aac"); //  128k aac
//     audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.mp3"); //  128k mp3
 //   audio.connecttohost("http://icecast.radiofrance.fr/franceinter-midfi.mp3"); //  128k mp3
//    audio.connecttohost("http://media.ndr.de/download/podcasts/podcast4161/AU-20190404-0844-1700.mp3"); // podcast
}

void loop()
{
    audio.loop();
    if (eos)
    {
    audio.connecttospeech("Wir funktionieren automatik Jetzt wollen wir tanzen mechanik", "de");
    eos=0;
    }
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreaminfo(const char *info){
    Serial.print("streaminfo  ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
    Serial.print("eof_speech  ");Serial.println(info);
    eos=1;
}
```

schreibfaul1, 2021-11-27:
Hello deladriere,
error found, the DC voltage offset has been eliminated

deladriere, 2021-11-28:
Hallo Wolle
Thanks this worked !

closed 2021-11-28 by deladriere