schreibfaul1/ESP32-audioI2S #232: Compile Errors on ESP32-S2

ajacobs-github, 2021-11-27:
Arduino: 1.8.16 (Windows 10), Board: "ESP32S2 Dev Module, Disabled, Disabled, Disabled, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from C:\Users\Alan\Documents\Arduino\Audio_Wifi_I2S_v1\Audio_Wifi_I2S_v1.ino:18:

C:\Users\Alan\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:151:37: error: 'i2s_dac_mode_t' has not been declared

     Audio(bool internalDAC = false, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN); // #99

                                     ^~~~~~~~~~~~~~

C:\Users\Alan\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:187:64: error: 'i2s_dac_mode_t' has not been declared

     [[deprecated]]void setInternalDAC(bool internalDAC = true, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN);

                                                                ^~~~~~~~~~~~~~

C:\Users\Alan\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:462:5: error: 'i2s_dac_mode_t' does not name a type; did you mean 'i2s_mode_t'?

     i2s_dac_mode_t  m_f_channelEnabled = I2S_DAC_CHANNEL_LEFT_EN;  // internal DAC on GPIO26 for M5StickC/Plus

     ^~~~~~~~~~~~~~

     i2s_mode_t

C:\Users\Alan\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:151:69: error: 'I2S_DAC_CHANNEL_LEFT_EN' was not declared in this scope

     Audio(bool internalDAC = false, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN); // #99

                                                                     ^~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Alan\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:151:69: note: suggested alternative: 'I2S_DATA_ENABLE_TEST_EN'

     Audio(bool internalDAC = false, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN); // #99

                                                                     ^~~~~~~~~~~~~~~~~~~~~~~

                                                                     I2S_DATA_ENABLE_TEST_EN

C:\Users\Alan\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:187:96: error: 'I2S_DAC_CHANNEL_LEFT_EN' was not declared in this scope

     [[deprecated]]void setInternalDAC(bool internalDAC = true, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN);

                                                                                                ^~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Alan\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:187:96: note: suggested alternative: 'I2S_DATA_ENABLE_TEST_EN'

     [[deprecated]]void setInternalDAC(bool internalDAC = true, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN);

                                                                                                ^~~~~~~~~~~~~~~~~~~~~~~

                                                                                                I2S_DATA_ENABLE_TEST_EN

Multiple libraries were found for "SD.h"

 Used: C:\Users\Alan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\SD

 Not used: C:\Users\Alan\Documents\Arduino\libraries\SD

exit status 1

Error compiling for board ESP32S2 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

schreibfaul1, 2021-11-28:
This library would not work reliably with the ESP32-S2. The ESP32-S2 is not powerful enough.

ajacobs-github, 2021-11-28:
Ok, thanks.

closed 2021-11-28 by ajacobs-github