schreibfaul1/ESP32-audioI2S #234: audio_showstreamtitle isn't working that well

JonLit, 2021-11-28:
I made a Webradio alarm clock, which (atm) should write the title of the song to the serial monitor. For testing I used SWR3 (Stream URL: https://liveradio.swr.de/sw282p3/swr3/play.mp3). If i play the stream with mpv in the terminal, I get this output (stripped):
`mpv --no-video https://liveradio.swr.de/sw282p3/swr3/play.mp3
(+) Audio --aid=1 (mp3 2ch 48000Hz)
AO: [pulse] 48000Hz stereo 2ch float
A: 00:00:01 / 00:00:04 (25%) Cache: 2.7s/99KB
File tags:
icy-title: Sky and sand / Paul Kalkbrenner feat. Fritz Kalkbrenner
A: 00:00:43 / 00:00:50 (88%) Cache: 5.8s/210KB
File tags:
icy-title: Mehr Hits mehr Kicks - einfach SWR3
A: 00:01:03 / 00:01:09 (91%) Cache: 5.9s/213KB
File tags:
icy-title: Sky and sand / Paul Kalkbrenner feat. Fritz Kalkbrenner
A: 00:02:03 / 00:02:09 (95%) Cache: 5.6s/205KB
File tags:
icy-title: SWR3 wünscht einen schönen Sonntag
A: 00:02:23 / 00:02:29 (96%) Cache: 5.9s/213KB
File tags:
icy-title: Sky and sand / Paul Kalkbrenner feat. Fritz Kalkbrenner`

If I play the same stream with my program I get only rarely a line from audio_showstreamtitle, which is also outdated, (2-3min old)
[RadioWecker.ino](https://github.com/schreibfaul1/ESP32-audioI2S/files/7614479/RadioWecker.txt)
[WebRadioWeckerWebsite.h](https://github.com/schreibfaul1/ESP32-audioI2S/files/7614480/WebRadioWeckerWebsite.txt)

schreibfaul1, 2021-11-29:
I haven't tested your program, mvp and audio_showstreamtitle() give the same results

mpv:
[image]

showstreamtiltle()
[image]

the difference would be: if the same string is always transmitted in the metadata for StreamTitle, then showstreamtitle() is only called once

JonLit, 2021-11-29:
Hmmm, but why does it not work for me? I've attached the serial output with comments (WiFi Network names and IPs were censored manually)

JonLit, 2021-11-30:
Turns out I'm stupid and used delay(1000) in my void loop()............

closed 2021-11-30 by JonLit