schreibfaul1/ESP32-audioI2S #236: Stuttering

cekay-amarda, 2021-12-06:
Have set psram on, 3mb no ota.
i also tried this with psram off and got same result.
Some songs will play one time with out a hitch, then stutter another time.  
Can't seem to figure this out as it happens at random. Same thing happens on all my devices. 
I use an attached rtos process for the playlist managment but the actual audio.loop() is called by loop. 
I also tried moving this to an rtos process with a priority of 3 and it makes no difference.
Any ideas?

cekay-amarda, 2021-12-06:
Actually, i lie. the audio.loop() does not run outside of loop. Will not run under attached process.

biologist79, 2021-12-06:
Probably this helps:
my project (ESPuino) makes use of Wolle's audiolib and it's running as a task: 
https://github.com/biologist79/ESPuino/blob/master/src/AudioPlayer.cpp#L254

What's especially important when doing so, is "wasting" time if no track is played as otherwise it's an infinite loop that will cause a restart: https://github.com/biologist79/ESPuino/blob/master/src/AudioPlayer.cpp#L738

cekay-amarda, 2021-12-06:
Thanks for answering, i am running player as one task and playlist management as completely different task. (Thanx for the help, i saw what i was doing wrong there), but still having stuttering issues. I've tried adding adding xtaskdelay into player loop and doesn't help

biologist79, 2021-12-06:
Had some issues with jumperwires. Did you solder all your stuff? If not I'd recommend doing so.
In my case webstream was fine but playing from SD was not.

cekay-amarda, 2021-12-06:
yes i did

cekay-amarda, 2021-12-06:
it comes and goes to which is wierd

schreibfaul1, 2021-12-07:
As biologist79 has already mentioned: all plug-in contacts must be reliable. The assemblies are arranged in such a way that the connections are as short as possible. The operating voltage must be constant over time. With large amounts of data, PSRAM is slower than the SRAM.
It is worth taking a look at the ESPuino project https://github.com/biologist79/ESPuino. There the audio library runs in a separate task. This can serve as an example for your own developments

hansipete, 2022-03-11:
I also had stuttering, crackling or no sound at all, with 10cm jumper wires between a `Joy It NodeMCU ESP32` and the `Sparkfun MAX98357A` amplifier.

Used pins 25,26,27 for I2S. Even just touching the wires fixed it for me. For the breadboard assembly I ended up wrapping the three data wires with another one and connect that to ground as "shielding".

Anyway... seems way to fragile. 10cm wire should not be that much of an issue.

closed 2022-08-13 by stale[bot]