schreibfaul1/ESP32-audioI2S #239: Problem with internal DAC's

bastian187, 2021-12-07:
@schreibfaul1 
Hello,
I want to use the internal 8bit DAC's of the ESP32 with this library with the arduino IDE, to have a simple internet radio.
I used the "audio I2S" example and I added to the setup:
``  Audio audio(true, I2S_DAC_CHANNEL_BOTH_EN);
    audio.setInternalDAC(true);   ``
And I delted: `` audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);  ``
If I load this to my ESP32, the serial Monitor shows: 
"  ***ERROR*** A stack overflow in task loopTask has been detected.
abort() was called at PC 0x40088a60 on core 1  "
I connected earphones to the DAC pins and I hear nothing!
Could anyone help me?

schreibfaul1, 2021-12-07:
Hello @bastian187
``
Audio audio(true, I2S_DAC_CHANNEL_BOTH_EN);
``
is correct and
``
//    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
``
is commented out
then the analog audio signal at PIN 25 and 26 can be heard.
Unfortunately the internal DAC does not (yet) work with version 2.0.1, https://github.com/espressif/arduino-esp32/issues/5938
better to use the Arduino version 1.0.6

bastian187, 2021-12-07:
Hello @schreibfaul1 , thank you for your answer.
But how can I add the esp32 boards in this version?
Under preferences I can't paste the link for adding the esp32 boards!

schreibfaul1, 2021-12-07:
I don't know which IDE you are using.
For the Arduino IDE please have a look here,
https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/
in Boards Manager 'Select Version' and then select V1.0.6

FedericoBusero, 2021-12-07:
I2S_COMM_FORMAT_STAND_I2S -> I2S_COMM_FORMAT_STAND_MSB ?

schreibfaul1, 2021-12-07:
unfortunately not, it just comes:
Audio.cpp:165:69: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]

V2.0.0-rc2 is the last version that works with the internal DAC

bastian187, 2021-12-08:
[image]
I had already the ESP32 version V1.0.6. as you can see in the picture!

FedericoBusero, 2021-12-08:
@schreibfaul1 I2S_COMM_FORMAT_I2S_MSB is deprecated but I2S_COMM_FORMAT_STAND_MSB is the new one.

It did work (compile without warning and also work in reality) until V2.0.0-rc2 but indeed no longer in V2.0.1

So I guess you need to change code to I2S_COMM_FORMAT_STAND_MSB  in case of DAC builting playback in arduino core v2. Then, at least, it works in V2.0.0-rc2 and hope it will return

bastian187, 2021-12-08:
Hi,
I changed  I2S_COMM_FORMAT_STAND_I2S to I2S_COMM_FORMAT_STAND_MSB but I'm getting the same errors (stack overflow and no sound)!
Any ideas?

closed 2021-12-25 by bastian187