schreibfaul1/ESP32-audioI2S #240: Trouble with 3.5 inch RaspberryPi Display on ESP32-A1S

norbert-w, 2021-12-08:
Hi Wolle,
first of all - thank you for all the work you do in developing code and sharing it with the community!
I tried to run a 3.5 inch RaspberryPi display as you show in https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/Simple%20WiFI%20Radio/Simple_WiFi_Radio.jpg using the provided sketch (ILI9486 lib). 
With no success.
Rewriting the code for using a ILI9341 controlled Display the sketch works perfect.
So is there a chance to check whether the provided ILI9486 lib matches this display?
Thanks in advance,
Norbert.

schreibfaul1, 2021-12-08:
Hi Norbert,
please insert a picture or a link from your display here, maybe someone will recognize the display and can tell us the controller used.

norbert-w, 2021-12-08:
Hi Wolle,
thank you for your response. As you requested ... a photo of the display.

[image]

Btw. it works fine with a RaspberryPi so it certainly is not defect.
Regards,
Norbert.

schreibfaul1, 2021-12-09:
The display uses logic ICs. So the SPI frequency could be limited, maybe 20MHz?

norbert-w, 2021-12-09:
Thank you, Wolle, I'll try it this evening!

norbert-w, 2022-01-26:
Hi Wolle,
I finally got the display working. I made the connections by following this article
 [https://github.com/Bodmer/TFT_eSPI/issues/11](url) .
Your driver works fine - there is no issue - I'll close this thread.

closed 2022-01-26 by norbert-w