schreibfaul1/ESP32-audioI2S #241: Audio skips a bit when refreshing display

oculos, 2021-12-08:
Hi,

I have a TTGO T-Display, and this library works very well, except that when I update my display (TFT_eSPI), the audio skips for a fraction of second. 
Someone on Reddit told me I should use DMA and queues, but I have the impression that this library already does it.
Any tips on how I could optimize things so that the screen doesn't impact audio playback?

konkav-js, 2022-02-06:
Hey.

Did you figured this out?

I am trying to make an internet radio player with a spi display and i2s dac, and also have that glitch when updating the display. I think that the esp32 has enough memory to be able to buffer more, but the problem is, I do this purely from hobby, so I have not the slightest clue what to write, and where :) Can we even allocate an arbitrary amount of memory for DMA, or is there some hardware constraints?

I mean, it looks like if I update just a few characters at time, then I can't hear this glitch, so I could work out updating the display one char at time, but it would be better to have bigger dma buffer, so I can do other things too (I would like to make it set the clock from NTP, and download the weather info).

I hope the developer can drop a line or two about this.

Andr1961, 2022-03-19:
I update one letter at a time, otherwise
the sound is interrupted
https://youtu.be/rt5bk5OrWQA?t=23

schreibfaul1, 2022-03-19:
Just saw Andr1961's video, nice project with good graphics.
The audio lib must be constantly 'fed', even when drawing on a display. You need a separate task for this. Take a look at my radio project https://github.com/schreibfaul1/ESP32-MiniWebRadio it is drawn on the display without sound interruptions.

Andr1961, 2022-03-19:
Hi, MiniWebRadio has changed a lot, added audioTask,
 displays (I will try 2 displays since the library allows you to choose the type of display and gpio),
when I last watched your project there was one display, thanks for the hint I will try and of course thanks for the excellent library))

Andr1961, 2022-04-11:
> it is drawn on the display without sound interruptions.

Hi Wolle, I tried with my project, everything is super, even the photo shows without delay, thanks for the hint. It is possible in the future to add audio.getCodecname() and audio.getBitRate() to VS1053 for compatibility, if possible, thank you.

closed 2022-08-13 by stale[bot]