schreibfaul1/ESP32-audioI2S #242: Playing mp3 files in a function on ESP32

HeikkiHietala, 2021-12-10:
Hi,

I wanted to thank you for building this library. I am hoping to build a talking clock, with MP3 files for the sounds of "oh one"... "oh nine" and then "one" to "fifty nine". I have the files made already and they all fit under 2MB of SPIFFS memory on the ESP32.

What I am struggling with is trying to build a method for running a function that actually plays the sound. The example has only one sound file, and while I can play my own sound, adding more than one doesn't work.

So, what should the structure be in the loop that would allow for a sound to be played and then wait for the next?

schreibfaul1, 2021-12-12:
When an mp3 file has played to the end, the void audio_eof_mp3() event is triggered. A list of the files still to be played can be stored there.

HeikkiHietala, 2021-12-13:
So, it'd be possible to say something like 
while (!audio_eof_mp3())
{
audio.loop()
}

Also, does the mp3 object have to be deleted after it has played out, or is it deleted automatically?

Thank you so much for your kind assistance!

schreibfaul1, 2021-12-13:
You can set a flag in audio_eof_mp3() and query it in loop()
````
boolean eof = false;

void loop(){
    audio.loop();
    if(eof){
        eof = false;
        // do something
    }
}

void audio_eof_mp3(const char * info){
    eof = true;
}
````

HeikkiHietala, 2021-12-13:
I hope I am not getting on your nerves by now, but I am struggling to understand a couple of things:

1) why is the sound to be played in the setup as in audio.connecttoFS(SPIFFS, "/acquiring1.mp3");?
2) what is the proper way to get more than one sound to play?
3) do the mp3 objects need to be deleted somehow so as not to block all memory?

In essence, I would benefit greatly if you had the time to write a function that would take in the file name, play it, and return to wait for the next file? In pseudocode for the time 14.25:

play welcome message 1
play welcome message 2
wait for time to be acquired from NTP
loop
call playTime("14.mp3") to say the hour of the time
call playTime("25.mp3")
wait until minutes change
end loop

function playTime(String playtimefile)
  audio.connecttoFS(SPIFFS, playtimefile);
  audio.play()

Sorry to trouble you with this, but even getting the sound to play when it is not named in setup seems to be beyond my skills.

schreibfaul1, 2021-12-13:
I have added an example that you can use as a template for your own developments.
https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Talking_Clock

HeikkiHietala, 2021-12-13:
Well, this example takes care of all my cares.

Thank you so very much for your library and support! I will be sure to include you in the upcoming blog entry on this device I am building.

HeikkiHietala, 2021-12-14:
Hi again, 

I am back with some more issues (sorry for letting you think I was in the clear).

I was unable to actually try your clock due to no SD, and SPIFFS only accommodates less than what you have in the MP3s. 

When the clock starts, it should go through four mp3s: entering network, getting time, adjusting local time, and standing by. 

Then it would sit tight and every minute, if someone is close enough, it'd say the appropriate hour file, and minute file. 

I gathered from your example that the system goes to the time to be said with 
```
sprintf(chbuf, "/%03d.mp3", next_hour);
    audio.connecttoFS(SPIFFS, chbuf);
```
and that the chbuf has the file name in proper format. Then it loops back, and the time is said. At the end of the mp3 it goes to the void function and plays the appropriate ending for the time message.

In my version, I have this in the zsetup to play the first sound: 

```
  SPIFFS.begin();
  audio.connecttoFS(SPIFFS, "/hacking1.mp3");

```
and the eof looks like this:
```
void audio_eof_mp3(const char *info) { //end of file
  next_hour++;
  if (timefile > 0) {
    if (timefile == 1) {
      audio.connecttoFS(SPIFFS, "/acquiring1.mp3");      // stroke
      timefile++;

    }
    if (timefile == 2) {
      audio.connecttoFS(SPIFFS, "/standingby1.mp3");      // precisely
      timefile++;

    }
    if (timefile == 3) {
      audio.connecttoFS(SPIFFS, "/adjusting1.mp3");
      timefile++;

    }
  }
}

What happens here is that it plays the message in the setup, then jumps straight to the third message, even if I have seeded timefile as 1. 

If there a way of not having to look at th eend of the file in the function, but rather uust give it a file name and play it with audio.loop()?

I dearly hope I don't irritate you already.

schreibfaul1, 2021-12-14:
I set the variable timefile to 3 at the beginning of the first mp3 file. Then it would be timefile-- right?

HeikkiHietala, 2021-12-14:
What I meant was that the structure bypasses 1 and 2 to go straight to the third sound in the loop.

Is it possible to use audio.loop() elsewhere than in the main loop?

closed 2022-04-17 by stale[bot]