schreibfaul1/ESP32-audioI2S #243: stream preparation takes a lot of time

unregistered33, 2021-12-14:
As the code became more complex, the audio started playing with a long delay (about 3-5 seconds). The audio playback code has not changed, but a Web server (+ css / js / images) has been added. What could be causing the delay? Potentially out of memory problems?

Serial log:
info        PSRAM found, inputBufferSize: 283615 bytes
info        buffers freed, free Heap: 175256 bytes
info        Reading file: "/bells/castle1.mp3"
info        MP3Decoder has been initialized, free Heap: 151624 bytes
--------------------------------------------------------------------------> delay ~3-5 sec
info        stream ready
info        Content-Length: 111203
info        ID3 framesSize: 54

unregistered33, 2021-12-15:
It looks like magic. I started to turn off all the functionality (AsyncWebServer, ArduinoJSON, WiFi) step by step, and when I turned off WiFi, the delay disappeared. I thought I had found the problem, but when I plugged everything back in for an additional test, everything is also fine and works! I don't know how to explain this...

seife, 2022-01-23:
For me AsyncWebServer was not working well together with ESP32-audioI2S.
I thought I need it for AsyncElegantOTA until I found ElegantOTA which works without the AsyncWebServer.
I did not investigate further back then, unfortunately.

unregistered33, 2022-01-23:
Oh, no! It's work very well! I found problem - NTP synchronization via standart way. GetLocalTime() - blocks main loop and i have long delay. Now im using async ESPNTPClient (with a little modification for core 2.0.1) and all fine

seife, 2022-01-23:
Ok, then disregard my comment ;-)
I was seeing AsyncElegantOTA being unreliable when sound was playing at the same time, so switching away from it to ElegantOTA (and to normal webserver code) fixed that for me :-)

closed 2022-04-25 by stale[bot]