schreibfaul1/ESP32-audioI2S #247: Google TTS not working on short phrase

schmurtzm, 2021-12-29:
Hi, I use ESP32 devkit v4 with internal I2S DAC.
I remark that I have no output from connecttospeech function when I have a short sentence. I don't understand what happens but it seems it seems to be related to the number of syllables  : 

> "bonjour" -> no sound
> "bonjourno" -> no sound
> "bonjournot" -> sound OK
> "bonjour bonjour" -> sound OK

> "a", "ab", "abc" -> no sound
> "abcd" -> sound OK

😶‍

May be the sound is too short ?

Here the code that I'm using (based on I2Saudio example, just modified audio I2S to avoid saturation with internal DAC and modified serial input to use TTS).
<details>
<summary>TTS by serial input code</summary>

```c++
//**********************************************************************************************************
//*    audioI2S-- I2S audiodecoder for ESP32,                                                              *
//**********************************************************************************************************

#include "Arduino.h"
#include "WiFiMulti.h"
#include "Audio.h"
#include "SPI.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
//#define SD_CS          5
//#define SPI_MOSI      23
//#define SPI_MISO      19
//#define SPI_SCK       18
#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26

Audio audio(true, I2S_DAC_CHANNEL_BOTH_EN);
WiFiMulti wifiMulti;
String ssid =     "xxxxx";
String password = "xxxxx";

void setup() {
//    pinMode(SD_CS, OUTPUT);      digitalWrite(SD_CS, HIGH);
//    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
//    SPI.setFrequency(1000000);
    Serial.begin(115200);
//    SD.begin(SD_CS);
    WiFi.mode(WIFI_STA);
    wifiMulti.addAP(ssid.c_str(), password.c_str());
    wifiMulti.run();
    if(WiFi.status() != WL_CONNECTED){
        WiFi.disconnect(true);
        wifiMulti.run();
    }
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21); // 0...21

//    audio.connecttoFS(SD, "/320k_test.mp3");
//    audio.connecttoFS(SD, "test.wav");
//    audio.connecttohost("http://www.wdr.de/wdrlive/media/einslive.m3u");
//    audio.connecttohost("http://macslons-irish-pub-radio.com/media.asx");
//    audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.aac"); //  128k aac
//    audio.connecttohost("http://audio-edge-ey5nr.ams.s.radiomast.io/radioblues-flac"); //  128k mp3
//    audio.connecttohost("http://radios.rtbf.be/classic21-64.aac"); //  aac

    
    audio.connecttospeech("Bonjour je suis le petit truc qui essaie de parler.", "fr");
    // audio.connecttospeech("Wenn die Hunde schlafen, kann der Wolf gut Schafe stehlen.", "de");
}

void loop()
{
    audio.loop();
    if(Serial.available()){ // put streamURL in serial monitor
        audio.stopSong();
        String r=Serial.readString(); r.trim();
       // if(r.length()>5) audio.connecttohost(r.c_str());
        if(r.length()>0) audio.connecttospeech(r.c_str(),"fr");
        log_i("free heap=%i", ESP.getFreeHeap());
    }
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
    Serial.print("eof_speech  ");Serial.println(info);
}
```
</details>  

Thanks for this great library !

schreibfaul1, 2021-12-30:
Nobody noticed that before, thank you, the problem is now solved

schmurtzm, 2021-12-30:
Awesome fix speed ! Tested, it works !
Thanks for that 😄

Just by curiosity, do you know why playing  [a google TTS url](http://translate.google.com/translate_tts?ie=UTF-8&client=tw-ob&tl=en&q=bonjour) like an mp3 stream instead using your function gives buffering problem at the end of playback ?

schreibfaul1, 2021-12-30:
The stream that GoogleTTS returns is treated like all web files. These are mostly podcasts or songs on the Internet and therefore longer than 1..2 seconds. The mistake 'destroyed' a short piece at the beginning

schmurtzm, 2021-12-30:
Thanks a lot, this one can be closed ;)

closed 2021-12-30 by schmurtzm