schreibfaul1/ESP32-audioI2S #248: [Question] - Is it possible to make ESP8266SAM compatible with ESP32-audioI2S ?

schmurtzm, 2021-12-30:
Hi, almost all is in the title.
Sam voice has a bad pronunciation quality compared to Google TTS and it's english only ... But it can work offline !

The idea is to use an ESP32 for home automation notifications. I was thinking that if my internet connection is down it could be useful to keep possible to send TTS notifications without Google TTS.

I took an eye in this ESP8266SAM library and the audio stream seems to be generated around [this line](https://github.com/earlephilhower/ESP8266SAM/blob/master/src/ESP8266SAM.cpp#L83).

Seems possible, isn't it ?

schreibfaul1, 2021-12-30:
The voice quality is not convincing. See beginning of https://www.youtube.com/watch?v=4ig14PAZ2Dw. There is a project for flite in my repository https://github.com/schreibfaul1/ESP32-flite-Arduino The quality is also bad.

schmurtzm, 2021-12-30:
Yes it's not good... but it work offline which can be useful in some case.
Awesome, you have already tested it ! You make a lot of things ;) 
Why your flite project is not based on ESP32-audioI2S ? Flite doesn't generate a classical wav ?

closed 2022-04-17 by stale[bot]