schreibfaul1/ESP32-audioI2S #249: There will always be background noise！

livebelive, 2022-01-02:
There will always be background noise in standby mode! I don't know if it's my equipment problem or where it's not set correctly, or it should be like this.
I tried to disconnect the BCLK pin and the noise disappeared.
But disconnecting dout and LRC doesn't work.
How can I completely remove the background noise?
Or is there any way to completely turn off the I2S channel output after each playback? I think the problem is that I2S is not turned off, but I'm not sure whether the device is allowed to turn off I2S.
My hardware is esp-wroom-32s

By the way, I changed the loop from void to bool. In my case, I built it with esphome. What I want to achieve is to register the "on_play" and "on_stop" services in homeassistant, so that I can use ha to call TTS and play music.

But strangely, when I judge the playback status, the playback is always not smooth. No matter how I set the conditions, there will be various problems. Finally, after I changed the loop from void to bool, the playback is smooth
The following are my changes. So far, no adverse effects have been found!

`    Audio.h:bool loop();
    Audio.cpp:
bool Audio::loop() {

    // - localfile - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
    if(m_f_localfile) {                                      // Playing file fron SPIFFS or SD?
        processLocalFile();
    }
    // - webstream - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
    if(m_f_webstream) {                                      // Playing file from URL?
        if(!m_f_running) return false;

//        if(m_f_m3u8wait){                                    // timer, wait for fresh m3u8 playlist
//            if(m_t1 < millis()){
//                log_i("wait zurückgesetzt");
//                m_f_m3u8wait = false;
//                processPlayListData();
//            }
//        }
        if(m_datamode == AUDIO_PLAYLISTINIT || m_datamode == AUDIO_PLAYLISTHEADER || m_datamode == AUDIO_PLAYLISTDATA){
            processPlayListData();
            if(m_f_m3u8data) processWebStream();
            return true;
        }
        if(m_datamode == AUDIO_HEADER){
            processAudioHeaderData();
            if(m_f_m3u8data) processWebStream();
            return true;
        }
        if(m_datamode == AUDIO_DATA){
            processWebStream();
            return true;
        }
    }
    return false;
}`

Although it seems that there is no result in the loop condition, it can solve the problem of poor playback. This may be due to a problem with the esphone platform..

下面是我的项目源码：
`#include "esphome.h"
#include "Audio.h"
#include "Arduino.h"

#define I2S_DOUT      26
#define I2S_BCLK      27
#define I2S_LRC       25

Audio audio;

bool startPlay = false;
bool tryClosing = false;
bool isStop = false;
bool Playing = false;
const char * toPlay = nullptr;

// void audio_eof_stream(const char *info){
//   isStop = true;
// }
class ESPAudioComponent : public Component, public CustomAPIDevice, public BinarySensor {

  public:
    
    void setup() override {

        audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
        audio.setVolume(12);//0------>21
        audio.i2s_mclk_pin_select(0);
        register_service(&ESPAudioComponent::on_play_media, "play_media", {"url"});
        register_service(&ESPAudioComponent::on_stop_media, "stop_media");
        publish_state(false);
    }
    void loop() override {
        audio.loop();
        if (audio.isRunning())
        {
            if (!audio.loop() || tryClosing)
            {   
              ESP_LOGD("custom", "Closing");
              audio.stopSong();
              publish_state(false);
              tryClosing = false;
            }
        }
        if(tryClosing)
        {
          audio.stopSong();
          ESP_LOGD("custom", "Stopped");
          tryClosing = false;
        }
        else if(Playing)
        {
          Playing = false;
          String r=toPlay; r.trim();
          audio.connecttohost(r.c_str());
          ESP_LOGD("custom", "Playing...");
          publish_state(true);
          toPlay = nullptr;
        }
        else if(isStop)
        {
          isStop = false;
          audio.stopSong();
          publish_state(false);
          ESP_LOGD("custom", "Stopped");
        }   
    }

    void on_stop_media() {
      tryClosing = true;
      audio.stopSong();
    }
    void on_play_media(std::string url) {
      toPlay = url.c_str();
      audio.stopSong();
      Playing = true;
    }
};`

schreibfaul1, 2022-01-03:
I cannot contribute anything to esphome. I suspect a not completely stable operating voltage. Alternatively, this could help:
if (audio.isRunning()) {
    // turn on the amplifier
}
else {
    // turn off the amplifier
}

livebelive, 2022-01-03:
Thank you for your reply！
I don't understand “// turn on the amplifier”
amplifier mean I2S？
how to rurn on/off I2S？
Is there a template reference in this part ？

livebelive, 2022-01-03:
The working voltage of the equipment is OK
I use a 5V 3A power adapter to power my DAC and esp32

livebelive, 2022-01-03:
by the way，after the equipment is powered on, there will be 16000KHz signal input on the DAC side

FrankBoesing, 2022-01-03:
It is an hardware issue.

livebelive, 2022-01-03:
> 这是一个硬件问题。

Here's a new esp32. I'll try it

schreibfaul1, 2022-01-03:
I don't think the ESP32 is the problem. Maybe ESP32 ground is not sufficiently decoupled from signal ground

livebelive, 2022-01-03:
it means problem maybe in DAC?

FrankBoesing, 2022-01-03:
The wiring is the problem - or you ned to add some power supply/ audio filtering. Nothing that could be fixed in the libary.

livebelive, 2022-01-03:
If so, does other people use I2S DAC also have this problem？

livebelive, 2022-01-03:
Does this happen when you use pcm5102？
我使用的是 PCM5122 but itis not only pcm5122
This DAC is an integrated board for raspberry pi

livebelive, 2022-01-03:
[https://img.alicdn.com/imgextra/i3/93631139/O1CN01aibkkl1KHhmsfTcly_!!93631139.jpg](url)
This one

livebelive, 2022-01-03:
[https://melody.liveviva.cn:8123/local/DAC.jpg](url)

livebelive, 2022-01-03:
It should be the power problem, but I think everyone should encounter this problem. Do we have any methods to deal with it?

closed 2022-04-25 by stale[bot]