schreibfaul1/ESP32-audioI2S #253: separate_audiotask

norbert-w, 2022-01-04:
Hi Wolle,
I took parts of your new sketch and adopted it to an existing web radio program.
You did a great job, Wolle, most functions work like a charm, for example the volume control.
But I couldn't find a way yet to connect from the startup station to a different one.
case STATION_DOWN:  {
                                  if(cur_station>0) 
                                  {
                                    cur_station--;
                                  } else cur_station = max_stations - 1;
                                  write_stationNr(cur_station);
                                  memcpy(Stat_Text, stations[cur_station].sender_url, strlen(stations[cur_station].sender_url));
                                  audioConnecttohost(Stat_Text);

All I get is the error message   [E][Audio.cpp:4179] playSample(): Can't stuff any more in I2S...
The station is still the previous one - but the sound quality is poor.
It would be nice to get a tip how to solve the problem.

Thanks,
Norbert.

schreibfaul1, 2022-01-04:
Hello Norbert,
the example is still brand new and not quite finished. Putting the Lib in its own task has the advantage that 'stuttering' in time-critical routines is prevented, e.g. web server, large TFT graphic ...
I added an example of how a own audio task works in the 'Simple WiFi Radio'
https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/OwnAudioTask/ESP32_Simple_WiFi_Radio_audioTask.ino
greetings
Wolle

norbert-w, 2022-01-04:
Hello Wolle,
thank you for your fast response - and the new sketch.
To me your example is quite unique while "simple" and that's what makes it so special.
So it's not criticism but telling what I experienced.

I'll have a close look on the code again and will report tomorrow,
Norbert.

norbert-w, 2022-01-05:
Hi Wolle,
I found **my** mistake!
It was a "leftover" audio.loop() call in a subroutine, that I forgot to erase.
Your sketch (and my program) work fine.
Thanks again for your great work,
Norbert

closed 2022-01-05 by norbert-w

runo4938, 2023-12-09:
Hi Wolle,
Sorry for my English. How to get a signal for VU-meter for the display?