schreibfaul1/ESP32-audioI2S #256: Interrupted Playback from SD

jonascant, 2022-01-06:
I have tried using 3 different SD Cards, and two different ESP32 Modules (one WROOM and one Lolin D32 Pro WROVER). I used a generic SD Module from ebay aswell as the built in SD connector of the WROVER. I am playing through the PCM5102A DAC.
I also tried various bitrates and framerates aswell as formats(mp3, wav, flac).

In all cases my playback was constantly interrupted, as in playing a fraction of a second, then silence and repeat..

I noticed that lowering the bitrate (to 48 k) improves the playback, as in more sound and less silence.
Playing from the internet runs flawless and the quality is nice.

Do i need a special type of SD Card?
I am making an installation with 20 touch sensors. Touching one sensor should play a dedicated file from sd card, without any loading delay. Is that a reasonable task for the ESP32 Wrover with PSRAM?

schreibfaul1, 2022-01-07:
no special SD card is required.
The following points are important:
- the operating voltage must be sufficient and constant,
-SPI lines should be as short as possible
- Solder as much as possible, do not use a breadboard
-SPI frequency can be 80MHz
-CPU frequency must be 240MHz,
-Arduino V2.0.2 does not work
-PSRAM is not required for mp3
If the points are met, mp3 can also run at 320Kbit/s

tueddy, 2022-01-20:
> Arduino V2.0.2 does not work

Your library works fine with latest Arduino 2.0.2 master, tested with
ESP-IDF version: v4.4-beta1-308-gf3e0c8bc41

jonascant, 2022-01-21:
The Problem still persists. I made sure of all connections - they are all soldered. 
All the requirements mentioned are met.
I am only using the example code. Wifi Radio playback runs fine. SD card is not working (interrupted).
Arduino version: 1.8.19

jonascant, 2022-01-21:
This is my board:
[image]

jonascant, 2022-01-21:
just updated to 0.20.2 same problem.. 
Could it be related to the format of the Card? I am using fat32, formated with gparted.

schreibfaul1, 2022-01-21:
The version: 1.8.19 is unknown to me. I can recommend V1.0.6 as a stable version. There will be no contact problems with the SD card, the structure looks completely clean.

jonascant, 2022-01-21:
I can't find a version 1.0.6. I am on arch linux. The lowest version i can install is 1.8.9: 
```
    1)  arduino  1  1.8.9   1  remote
    2)  arduino  1  1.8.9   2  remote
    3)  arduino  1  1.8.9   3  remote
    4)  arduino  1  1.8.12  1  remote
    5)  arduino  1  1.8.13  1  remote
    6)  arduino  1  1.8.13  2  remote
    7)  arduino  1  1.8.13  3  remote
    8)  arduino  1  1.8.14  1  remote
    9)  arduino  1  1.8.15  1  remote
   10)  arduino  1  1.8.16  1  remote
   11)  arduino  1  1.8.17  1  remote
   12)  arduino  1  1.8.18  1  remote
+  13)  arduino  1  1.8.19  1  remote
+  14)  arduino  1  1.8.19  1  /var/cache/pacman/pkg
```

I also tried arduino-cli, version 0.20.2 and 0.19.3.
How can i go about ruling out issues? As written above, I already swaped every component. I guess it has to be the code, the file itself or the compiling options. I can also create a wiring Diagram of what I did.

biologist79, 2022-01-21:
Just to clearify:
1.0.6 => ESP32-Arduino-version
1.8.19 => Version of Arduino-IDE

biologist79, 2022-01-21:
@jonascant I'd recommend SDMMC instead of SPI along with this board: https://de.aliexpress.com/item/32867572635.html. Have been using this module many many times and it works fine.
Make sure to remove pullup-resistor for GPIO 2 (https://forum.espuino.de/uploads/default/optimized/1X/9350aa36bf55ad0b22d7e0d4ac2da0e31a7249c2_2_465x750.jpeg) and only connect this SD-module to 3.3 V as it doesn't provide a voltage-regulator. Problem with GPIO 2: you cannot flash ESP32 when SD-card is inserted (when SD is removed it's fine). My way to to pullup anyway is to configure GPIO 2 as INPUT_PULLUP: https://github.com/biologist79/ESPuino/blob/master/src/SdCard.cpp#L31. Works fine.

jonascant, 2022-01-21:
I am sorry, i was confused about the versions. Thanks for clarifying. I replaced my 2.0.2 ESP32-Arduino with 1.0.6 . It just worked immediately. Thanks!

closed 2022-01-21 by jonascant