schreibfaul1/ESP32-audioI2S #270: SD Card question.

nodoubtman, 2022-01-28:
Hello.
I have a question about the sd card.

What i have to do to use the default sd card format.

I have to use SDFat, but i have to list the files to play it through web browser.
Thanks.
Marc.

schreibfaul1, 2022-01-28:
With the ESP32, file names are read and written in ASCII (Basic Latin). I would only recommend SDFat if, for example, Cyrillic Chinese characters are involved.
Must be enabled with #define SDFATFS_USED in audio.h. There are different versions of SDFat, I think the beta is needed for the extended character set (in UTF-8).

nodoubtman, 2022-01-28:
Hello, thanks for the answer.

I have to list the mp3 on the sd card. place them in an struct array then play throught the web interface.
But it doesnt seem to work.

can you help?
Thnks.
Marc.

nodoubtman, 2022-01-29:
I fix my problem with Sd_Lib.h thanks.
Marc.

closed 2022-01-29 by nodoubtman