schreibfaul1/ESP32-audioI2S #271: SPIFFs with SDFat impossible

nodoubtman, 2022-01-30:
Hello... I would like to use spiffs with SDFat System, but i cannot... 
is it possible to do a work around?

Thanks.
Marc.

schreibfaul1, 2022-01-30:
That does not work. SDFat comes with its own file system. This is not compatible with that of Espressif. SPIFFS, SD, SD_MMC are therefore not possible.

nodoubtman, 2022-01-30:
> That does not work. SDFat comes with its own file system. This is not compatible with that of Espressif. SPIFFS, SD, SD_MMC are therefore not possible.

Thanks.
Can i get another sd card which is not sdhc that can work ?
Thanks.
Marc.

schreibfaul1, 2022-01-30:
I don't quite know what the question means. Regarding SDFat I have no idea.
A second instance can be created with the ESP32 libraries (SD1, SD2, ...) I think this must be enabled in the sdkconfig, the default is one instance.

nodoubtman, 2022-02-04:
ok

closed 2022-02-04 by nodoubtman