schreibfaul1/ESP32-audioI2S #274: connecttospeech not working

hexeguitar, 2022-02-03:
`bool Audio::connecttospeech(const char* speech, const char* lang)`
does not update the  `WiFiclient* _client` pointer to `client`, thus the received data is not processed in the loop.
Fix:
add
`_client = static_cast<WiFiClient*>(&client);`
to the body of the function,
Tested on LyraT 4.3 (ESP-Wrover-B)

schreibfaul1, 2022-02-04:
Thanks for the hint. I wanted to replace GoogleTTS with another service, too often something was changed by Google. But this time it was my fault. ConnectToSpeech works again!

schmurtzm, 2022-02-20:
It will be probably hard to find something as good and free than GoogleTTS... 
This is a major feature of your library, please keep it ;)

hansipete, 2022-03-13:
I had problems with texttospeech and could fix them by changing lines around 748 in Audio.cpp like so:
```
_client = static_cast<WiFiClientSecure*>(&clientsecure);
    if(!_client->connect(host, 443)) {
        log_e("Connection failed");
        return false;
    }
    _client->print(resp);
```

closed 2022-08-13 by stale[bot]