schreibfaul1/ESP32-audioI2S #275: getFilePos and PSRAM buffer

ceear, 2022-02-04:
If you call the function `getFilePos()` and psram is available, the call returns a value that seems to be too high.
I noticed this with quite short (60 seconds) MP3 files. If you save the position with `getFilePos()` and later restore it with `setFilePos()` you end up at a different point in the file. So you don't really start at the same position again.

If you disable psram, getFilePos() returns valid values.
Is it possible that there is another setting on my side?

Cheers,
Christian

biologist79, 2022-02-04:
Hi Christian,

noticed the same yesterday. Had small audiofiles and ESPuino's Neopixel indicated 3/24 LEDs upfront at the very beginning. For large files it was fine. But I didn't validate the values to be honest; just monitored percentage: https://github.com/biologist79/ESPuino/blob/master/src/AudioPlayer.cpp#L739.

Used PSRAM, too.

ceear, 2022-02-04:
Hi,

yes indeed also visible in the ESPuino progress bar.
getAudioCurrentTime() seems to give correct results though.

(and of course I noticed that in the context of ESPuino).

biologist79, 2022-02-04:
Should revise my code a bit at this point. Guess I added modulo-division in my code only for debugging-purposes (println) and forgot to remove it. By having milis() %100 it's a bit of a luck when the calculation is saved + doesn't make sense to do the first calculation if it's not needed because modulo-divsion isn't true :-) Will change that, it's "ausbaufähig" :-)
However, shouldn't be the issue here.

ceear, 2022-02-04:
> shouldn't be the issue here.

I agree. I adjusted this some time ago. the calculation is now done at regular intervals but the result remains the same ;-)
https://github.com/ceear/ESPuino/blob/a0f42c654a4baa9d73edba69ac11e58af251ac97/src/AudioPlayer.cpp#L730

schreibfaul1, 2022-02-04:
Hello everyone,
getFilePos() returns the read/write position of the file pointer at query time. What is heard at the moment may differ. That depends on how full the audio buffer is. In fact, the file pointer will often be a few seconds ahead of what is heard.

uint32_t just_heard;
just_heard = audio.getFilePos() - audio.inBufferFilled();

biologist79, 2022-02-04:
Oh, thanks for clarification @schreibfaul1. Will test this.

ceear, 2022-02-04:
makes sense and works great.
Thank you!

closed 2022-02-04 by ceear