schreibfaul1/ESP32-audioI2S #277: How to change web radio stations

marchingband, 2022-02-06:
Hello
I'd like to connect to a URL, then in response to a button press, connect to a different URL ... basically change the station on demand.
What is the correct way to do this? I am getting a lot of crashes, with using audio.stopSong() and then audio.connecttohost(URL).
Is it possible for this transition to be seamless? Ideally I'd like to keep playing the first station while the second one connects and buffers, and then quickly switch over.
thank you! loving this library!

nodoubtman, 2022-02-06:
// SLEEP RADIO
bool hostConnected =  audio.connecttohost("http://149.56.234.138:8169/stream");

Serial.println(hostConnected ? F("HOST CONNECTED") : F("CANNOT CONNECT TO HOST!") );

in the loop

void loop(){
 audio.loop();
}

marchingband, 2022-02-07:
thanks @nodoubtman I found my crash elsewhere.
Has anyone tried using an AsyncClient, or something like that, to buffer http streams before switching?

closed 2022-04-25 by stale[bot]