schreibfaul1/ESP32-audioI2S #285: Reseting when playing streams

nitko12, 2022-02-21:
HI,

I've been trying to use your library to stream aac/mp3 radio stations and after some time (around 10 minutes) of playing it crashes the esp.

Before the crash it outputs "Stream is slow" and upon reconnecting crashes.

I've added https://github.com/e-radionicacom/Soldered-ESP32-Internet-Radio-Kit/blob/3323a43a8269ac8bd55049e9fa77c4b1e0c62cfe/lib/ESP32-audioI2S/src/Audio.cpp#L3947 these lines which keep it from crashing as the issue seems to be losing wifi connection randomly, but this doesn't solve my issue as there is a pause in playback.

Have you experienced anything similiar or have tips on how to fix this?

I've tested on two stations, one aac, one mp3 and the issue persists.

The code is ran on our Dasduino board prototype with PSRAM enabled.

Here is the output and decoded stack trace:
```info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        Stream lost -> try new connection
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
info        buffers freed, free Heap: 184696 bytes
info        Connect to new host: "http://stream.otvoreni.hr/otvoreni"
info        Connect to "stream.otvoreni.hr" on port 80, extension "/otvoreni"
[I][WiFiClient.cpp:243] connect(): select returned due to timeout 250 ms for fd 55
info        Request http://stream.otvoreni.hr/otvoreni failed!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4015e99c  PS      : 0x00060030  A0      : 0x800dbddc  A1      : 0x3ffb1ef0  
A2      : 0x3ffc1bcc  A3      : 0x3f800024  A4      : 0x00001000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1e00  
A10     : 0x3ffbd1cc  A11     : 0x400892b4  A12     : 0x000000a0  A13     : 0x000000fa  
A14     : 0x00000a0d  A15     : 0x000a0d0a  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4008ca01  LEND    : 0x4008ca11  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x4015e99c:0x3ffb1ef0 0x400dbdd9:0x3ffb1f40 0x400dca63:0x3ffb1f70 0x400d16ce:0x3ffb1f90 0x400e9cfd:0x3ffb1fb0 0x4008e81e:0x3ffb1fd0
  #0  0x4015e99c:0x3ffb1e in WiFiClientRxBuffer::read(unsigned char*, unsigned int) at /Users/nitkonitkic/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.cpp:572
      (inlined by) WiFiClient::read(unsigned char*, unsigned int) at /Users/nitkonitkic/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.cpp:437```

t30shupp, 2022-05-01:
I mean, the problem comes from somewhere else. I have poor WiFi reception and I also had these problems. After I changed the timeouts (const uint32_t TIMEOUT_MS{1000}; const uint32_t TIMEOUT_MS_SSL{3600}; in audio.cpp) this problem occurs very rarely. Maybe in a future version you can make these times configurable.

schreibfaul1, 2022-05-04:
Thanks for the hint, I added the function `setConnectionTimeout(uint16_t timeout_ms, uint16_t timeout_ms_ssl)` to adjust the thresholds.

schuh8, 2022-05-17:
At last success! I first changed the timeouts to: (const uint32_t TIMEOUT_MS{1000}; const uint32_t TIMEOUT_MS_SSL{3600}; in audio.cpp) as suggested by t30shupp. Dropouts were far less but still frequent enough to be annoying. I then changed the values to 2000 and 7200 (not quite knowing what I was doing, but on the theory that if a little is better, more is better yet). Surprisingly it worked perfectly with no dropouts on any of the 10 radio streams I normally play over many many hours. Thanks schreibfaul1 & t39shupp.

schreibfaul1, 2022-05-17:
Timeout is the time to wait if the connection fails. You can also with ```` audio.setConnectionTimeout(timeout_ms, timeout_ms_ssl)  ```` change the time. The times I set are only empirical values.

t30shupp, 2022-05-17:
There may also be another reason. I am not sure. In my project, I followed the "OwnAudioTask" example and set the priority of the task to 2, as in the example. But my tests show me that 4 is the better value for me. Maybe you have to increase the timeout times for lower priority.

closed 2022-08-13 by stale[bot]