schreibfaul1/ESP32-audioI2S #286: No sound when using SD card

benedek0203, 2022-02-27:
Hi,

I tried to use this library, "connecttohost" is working flawlessly, but I haven't got luck with "connecttoFS" with SD card. I can read the SD card normally, so I suppose the issue is deeper somewhere.
Audioinfo output.:
PSRAM not found, inputBufferSize: 6399 bytes
buffers freed, free Heap: 151068 bytes
Reading file: "/test.mp3"
MP3Decoder has been initialized, free Heap: 127456 bytes
Content-Length: 9241756
ID3 framesSize: 462395
ID3 version: 2.3
ID3 normal frames
syncword not found
End of file "test.mp3"

As I see it stops with "syncword not found" but looking at your code I don't really know why.

NodeMCU ESP32 module, lib.: 2.0.2
MAX98537 (working okay with connecttohost)
Your lib is commit [058f132]

Thank you for your help.

kkuez, 2022-03-05:
I second that.

schreibfaul1, 2022-03-06:
The ID3 header of an mp3 file is read out (completely?) and processing stops at the audio block. Any other hints?

optimaxch, 2022-05-07:
from SD card workinf fine a  .mp3  when using arduino IDE,  but "scratching sound" after uploading with platformIO vs code.
on ESP32 pico kit

schreibfaul1, 2022-05-07:
difficult to say, these settings work for me

````
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

platform_packages =
    platformio/tool-esptoolpy @ ~1.30100
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
````

jasonrepos, 2022-05-08:
I can confirm that I also experienced "scratching / inaudible sound" when compiling using platformIO, however after changing my ini to the following as desribed by @schreibfaul1, it works great

[env:esp32dev]
#platform = espressif32
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
monitor_speed = 115200

closed 2022-08-13 by stale[bot]