schreibfaul1/ESP32-audioI2S #289: Audio File Duration Limit

CoretechR, 2022-03-20:
Hi, when using the function getAudioFileDuration() on a 19 hour long audiobook the duration was reported wrong.

It looks like the variable m_audioFileDuration is enough for 74 hours at 128kbit. But when it is multiplied by 8 in getAudioFileDuration() it overflows after only 9 hours.
`m_audioFileDuration = 8 * m_audioDataSize / m_avr_bitrate;`

Putting the division in brackets solved the problem for me:
`m_audioFileDuration = 8 * (m_audioDataSize / m_avr_bitrate);`

schreibfaul1, 2022-03-21:
thanks, i corrected it

CoretechR, 2022-04-04:
I found a few more spots where the integer limit has an effect.
For example:
```
if(m_avr_bitrate) m_audioCurrentTime = (pos-m_audioDataStart) * 8 / m_avr_bitrate;
```
This order of division/multiplication allows for longer files:
```
if(m_avr_bitrate) m_audioCurrentTime = (pos-m_audioDataStart) / m_avr_bitrate * 8;
```
And there are a few more spots where m_audioCurrentTime is updated. I applied this fix to all of them and tested it with my 19 hour file. 
I have not tried the function setAudioPlayPosition() but it might apply to it as well. Again, no real need to change this, it's only necessary for really long files.

Sorry if I'm being annoying. I'm just working on a little audibook player and testing the limits with long files. Without your library this would be an impossible amount of work.

Best regards,
Max

schreibfaul1, 2022-04-06:
Die Stellen sind jetzt so wie in Deinem Vorschlag geändert. Hoffe, das funktioniert nun auch mit der 19h Datei.
vG
Wolle

CoretechR, 2022-04-06:
Ich hab es gerade getestet, funktioniert perfekt. Vielen Dank!