schreibfaul1/ESP32-audioI2S #293: Updating m_resumeFilePos after resuming file

CoretechR, 2022-04-03:
Hi,

for me, m_audioCurrentTime does not update correctly when opening an mp3 file with the parameter m_resumeFilePos. If looks like you thought of that already in processLocalFile():
```
if(!f_stream && m_controlCounter == 100) {
        f_stream = true;
        if(audio_info) audio_info("stream ready");
        if(m_resumeFilePos){
            if(m_resumeFilePos < m_audioDataStart) m_resumeFilePos = m_audioDataStart;
            if(m_avr_bitrate) m_audioCurrentTime = (m_resumeFilePos - m_audioDataStart) * 8 / m_avr_bitrate;
            audiofile.seek(m_resumeFilePos);
            InBuff.resetBuffer();
            log_i("m_resumeFilePos %i", m_resumeFilePos);
        }
    }
```

Maybe this does not work if the bitrate is not ready by the time the if-condition runs and the f_stream flag is reset, so updating m_audioCurrentTime is skipped.
As a quick fix, I created another flag (f_resume) that gets reset only after the bitrate was set. This seems to have solved the problem for me.
```
if(!f_resume && m_resumeFilePos){
    if(m_resumeFilePos < m_audioDataStart) m_resumeFilePos = m_audioDataStart;
    if(m_avr_bitrate){
        m_audioCurrentTime = (m_resumeFilePos - m_audioDataStart) * 8 / m_avr_bitrate;
        f_resume = true; 
    }     
}
```

schreibfaul1, 2022-04-04:
Of course, if connecttoFS() is called with resumeFilePos>0, then the m_avr_bitrate is not yet known and has the value 0. Division by zero would then lead to an error. I found another way to calculate the correct m_audioCurrentTime.

Best regards
Wolle

CoretechR, 2022-04-04:
That works perfectly for me.
Vielen Dank, Wolle!

closed 2022-04-04 by CoretechR