schreibfaul1/ESP32-audioI2S #294: PSRAM Size and Failure to Connect

Quaestuosa, 2022-04-08:
Hello,

as suggested here (https://github.com/schreibfaul1/ESP32-audioI2S/issues/290) I've purchased a different module now that has PSRAM. It is a ESP32 Wrover with 8 MB PSRAM according to the datasheet.
Now everything compiles fine and the applications plays back the selected audio stream. What a great software! It boots and connects much faster and seems to be more stable than my previous web radio using a Raspberry Pi.

However, I get the message _> PSRAM found, inputBufferSize: 283615 bytes <_ that only about 300k bytes PSRAM are used. In the source code of _audio.h_  it says that most stations only allow buffering up to that size. I'm not sure what kind of buffering my favorite radio stations offer. But wouldn't it be possible or reasonable to use the max. available size of PSRAM for buffering?

Even with an external antenna connected to the module I often have poor Wifi signal quality resulting in messages like _< slow stream, dropouts are possible <_. Maybe a larger buffer would eliminate this - if supported by the radio station.

Will the software automatically try to reconnect if the Wifi connection is lost or do I have to check and reconnect myself in the main loop()?

And one more issue:
When starting up sometimes I get this message:
_
info        Connect to "streams.br.de" on port 80, extension "/b5aktuell_2.m3u"
info        Request http://streams.br.de/b5aktuell_2.m3u failed!
_

Then I have to reboot the module to start the radio stream. Is there a way to detect the failure to connect in the main loop() and somehow (how?) try to connect manually?

Thank you very much!
Cheers
Markus

schreibfaul1, 2022-04-09:
Hello,
most stations broadcast less than 100k in advance, some 200k. Thus, a larger audio memory would not be completely filled. If more data is read than written, the message 'slow stream, dropouts are possible' appears from a lower memory threshold. The threshold is slightly more than a frame size (for aac or mp3 that's 1600 bytes). If there is no data for several seconds, an attempt is made to reestablish the (TCP) connection. That doesn't help against a WiFi interruption or if the connection to the Internet is somehow disrupted, that would have to be recognized in the main program.

http://streams.br.de/b5aktuell_2.m3u is a playlist. BR24 works for me. Alternatively, you could call up the stream directly. http://dispatcher.rndfnk.com/br/br24/live/mp3/mid maybe that works better.

best regards
Wolle

Quaestuosa, 2022-04-09:
Hello,
Thank you very much for this detailed explanation. I'm really impressed and thankful that you take the time to patiently answer questions of users.
I've often seen this message (_slow stream_) and now understand what it means.

This afternoon I've done a lot of testing with different radio stations. Usually the connection is establish very quickly and then the data is streamed.
However there are occations that produce the previously mentioned error message
_Request http://radioStationX failed!_
Also I got this message
_info        Connection has been established in 74 ms, free Heap: 219736 bytes
info        HTTP/1.0 403 Account already in use_

In both occations a hardware reset solved the problem. But of course this ist not very user friendly (I'm doing this project for an elderly person). You mention that this (at least the first error message) could be resolved in the main program. Maybe also the second one?

Could you please give me some hints on how to detect if the radio stream is being played back and if not so how to reconnect to the stream? Maybe there's a possibility to see if the stream is playing? If not I simply could restart the module by software.

Thanks again and have a nice weekend!

Cheers
Markus

schreibfaul1, 2022-04-09:
Hi Markus,
usually the WiFi connection is stable and 24/7 operation is possible without interruptions. So I suspect a HW error. If you can rule out poor reception conditions, please use another ESP32 board. I have one that has an antenna but is not connected. It is therefore absolutely unstable.

Quaestuosa, 2022-05-04:
Hello again,
to catch the described problem I've added this code in the main() loop:
```
    if (audio.isRunning() == false)
    {
        audio.connecttohost(urlRadioStation);
    }

```
I've also tried to extend this condition to
`if ((audio.isRunning() == false) || (audio.inBufferFilled() == 0))
`
but this doesn't work because at startup the buffer always is empty.

After a lot of testing (> 1 week) I can report that most of the time (95%) the software starts up nicely. Once it is running after the initial startup it runs absolutely stable even with a weak Wifi signal.

Again thanks for the great work! For this application the ESP32 is a much better choice than the Raspberry Pi.

Cheers
Markus

closed 2022-05-04 by Quaestuosa