schreibfaul1/ESP32-audioI2S #301: Compile error with latest PIO 4.2.0 & Arduino master

tueddy, 2022-05-03:
Hi Wolle,

trying to compile ESPuino with PlatformIO 4.2.0 (just released and available via PIO update) & Arduino master results in compiler error:

`.pio/libdeps/lolin_d32_pro_sdmmc_pe/ESP32-audioI2S-master/src/Audio.cpp:216:18: error: 'TAG' was not declared in this scope
         ESP_LOGE(TAG, "Audio::setBufsize must not be called after audio is initialized");
                  ^~~
`

Adding this line in Audio.cpp before line 216 resolves the issue:
`static const char* TAG = "MyShinyTagAudioModule";
`

platform.ini:

`[env]
platform = espressif32
platform_packages =
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
`

Maybe you could this into account for your next version?

Best regards
tueddy

schreibfaul1, 2022-05-03:
Hello tueddy,
I replaced ESP_LOGE with log_e. This eliminates the 'TAG'. Hope that helps.

best regards
Wolle

tueddy, 2022-05-04:
Wolle, thank's for quick fix!
Compiles on all versions now..

closed 2022-05-04 by tueddy