schreibfaul1/ESP32-audioI2S #302: 8bit DAC's of the ESP32

Fire-Dragon-TW, 2022-05-03:
How can I use 8bit DAC's of the ESP32

Fire-Dragon-TW, 2022-05-03:
```C
#include "Audio.h"
#include "FS.h"
#include "SPIFFS.h"

#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26

Audio audio(true, I2S_DAC_CHANNEL_BOTH_EN);

void setup() {
    Serial.begin(115200);
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21);

    SPIFFS.begin();
    audio.connecttoFS(SPIFFS, "alert.mp3");
}

void loop()
{
    audio.loop();
}
```

Fire-Dragon-TW, 2022-05-03:
[image]
[image]

Fire-Dragon-TW, 2022-05-03:
speaker is connected to GPIO 26 and GND

ggssantos, 2022-05-08:
@Fire-Dragon-TW, this issue is not related to the library itself, it is a problem with the Espressif Arduino Core. The last version that the internal DAC works is the 2.0.0.
You need to open the board manager on your arduino IDE and downgrade the ESP32 to 2.0.0.

[image]

The issue was discussed here: https://github.com/espressif/arduino-esp32/issues/5938

jitendra1607, 2022-06-20:
@Fire-Dragon-TW @ggssantos 

I want to implement this library because it enables the internal DAC of ESP32 without any other hardware. Can you post a sample audio output recording, to judge the quality of simple speech voice? It would be really helpful to make a final decision. Thank you

closed 2023-04-14 by stale[bot]