schreibfaul1/ESP32-audioI2S #303: Volume Control

Quaestuosa, 2022-05-11:
Hello,
I'm using an external DAC based on the UDA1334A which should produce line level output. However the volume is quite low even when setting the volume to max using
`setVolume(21);`

Is there another way to increase the output level by software? To be honest, I don't understand how the setVolume() function works internally. But would it be possible to change the volumetable[] to increase the volume?
```
const uint8_t volumetable[22]={   0,  1,  2,  3,  4 , 6 , 8, 10, 12, 14, 17,
                                     20, 23, 27, 30 ,34, 38, 43 ,48, 52, 58, 64};

```

A forum post (don't remember where I found this) suggests to do a bit shift of the I2S data to increase volume. Would that be possible here?

If there's no way to do this in software I'd have to install a small amplifier board that delivers line level signals in stereo quality. Are there recommendations?

Thank you very much!

Cheers
Markus

biologist79, 2022-05-11:
Hi Markus,

ESPuino makes use of https://github.com/schreibfaul1/ESP32-audioI2S. That being said the only way (*) to change volume is setVolume() along with values 0-21. Internally values 0-21 are mapped to those ones you described: https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.h#L389.

To be honest: I've never been using UDA (without amp) attached to ESPuino only for lineOut. But what I've running is an ESP32 along with https://github.com/pschatzmann/ESP32-A2DP and UDA1334 connected to a mixing console (acting as a2dp-sink). This is fine so far but probably it's also at a very low level (and I just didn't notice because of the mixing console).

However, for my [Kopfhörerplatine](https://forum.espuino.de/t/kopfhoererplatine-basierend-auf-uda1334-pj306b/705) I'm using TDA1308 as amp. This setup also works fine as lineout.

(*) For AC101 there's another way.

Quaestuosa, 2022-05-11:
Hello,
Thank you very much for your detailled reply. I understand that there's no way to increase the volume by software. So I will add a small amplifier module to my project. An amplifier module for headphones is a good idea.
Cheers
Markus

closed 2022-05-11 by Quaestuosa

schreibfaul1, 2022-05-11:
Hello Quaestuosa,
the UDA1334A will probably be installed on an Adafruit board. I have one, the output voltage is very low compared to other DACs.
The volumetab only simulates a type of aurally correct volume control, such as potentiometers with a logarithmic characteristic are used in analog technology.
With audio.setTone(6, 6, 6); // this is the equalizer function ,low, medium, high +6dB
you can use the digital filters to amplify the signal, but accept non-linearities.
The best way has already been described by biologist79, that is to use the "Kopfhörerpatine", a headphone board, i.e. an analogue amplifier.